MagicMirrorOrg/MagicMirror #1050: Unable to install dependencies on installation

twalenzyk, 2017-10-08:
I'm getting this output on running mentioned installation:
```
pi@raspberrypi:~ $ bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"

$$\      $$\                     $$\           $$\      $$\ $$\                                          $$$$$$\
$$$\    $$$ |                    \__|          $$$\    $$$ |\__|                                        $$  __$$\
$$$$\  $$$$ | $$$$$$\   $$$$$$\  $$\  $$$$$$$\ $$$$\  $$$$ |$$\  $$$$$$\   $$$$$$\   $$$$$$\   $$$$$$\  \__/  $$ |
$$\$$\$$ $$ | \____$$\ $$  __$$\ $$ |$$  _____|$$\$$\$$ $$ |$$ |$$  __$$\ $$  __$$\ $$  __$$\ $$  __$$\  $$$$$$  |
$$ \$$$  $$ | $$$$$$$ |$$ /  $$ |$$ |$$ /      $$ \$$$  $$ |$$ |$$ |  \__|$$ |  \__|$$ /  $$ |$$ |  \__|$$  ____/
$$ |\$  /$$ |$$  __$$ |$$ |  $$ |$$ |$$ |      $$ |\$  /$$ |$$ |$$ |      $$ |      $$ |  $$ |$$ |      $$ |
$$ | \_/ $$ |\$$$$$$$ |\$$$$$$$ |$$ |\$$$$$$$\ $$ | \_/ $$ |$$ |$$ |      $$ |      \$$$$$$  |$$ |      $$$$$$$$\
\__|     \__| \_______| \____$$ |\__| \_______|\__|     \__|\__|\__|      \__|       \______/ \__|      \________|
                       $$\   $$ |
                       \$$$$$$  |
                        \______/

Updating packages ...
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists... Done                      
Installing helper tools ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.3).
unzip is already the newest version (6.0-21).
wget is already the newest version (1.18-5).
The following additional packages will be installed:
  libcurl3
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-arch git-cvs git-mediawiki git-svn
The following packages will be upgraded:
  curl git libcurl3
3 upgraded, 0 newly installed, 0 to remove and 61 not upgraded.
Need to get 3861 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf git armhf 1:2.11.0-3+deb9u2 [3380 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf curl armhf 7.52.1-5+deb9u1 [220 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libcurl3 armhf 7.52.1-5+deb9u1 [261 kB]
Fetched 3861 kB in 2s (1511 kB/s)
apt-listchanges: Can't set locale; make sure $LC_* and $LANG are correct!
Reading changelogs... Done
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
(Reading database ... 122683 files and directories currently installed.)
Preparing to unpack .../git_1%3a2.11.0-3+deb9u2_armhf.deb ...
Unpacking git (1:2.11.0-3+deb9u2) over (1:2.11.0-3) ...
Preparing to unpack .../curl_7.52.1-5+deb9u1_armhf.deb ...
Unpacking curl (7.52.1-5+deb9u1) over (7.52.1-5) ...
Preparing to unpack .../libcurl3_7.52.1-5+deb9u1_armhf.deb ...
Unpacking libcurl3:armhf (7.52.1-5+deb9u1) over (7.52.1-5) ...
Setting up libcurl3:armhf (7.52.1-5+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up git (1:2.11.0-3+deb9u2) ...
Setting up curl (7.52.1-5+deb9u1) ...
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v5.1.0
Installed Node version: v4.8.2
Node should be upgraded.
Installing Node.js ...

## Installing the NodeSource Node.js v6.x repo...

## Populating apt-get cache...

+ apt-get update
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done

## Confirming "stretch" is supported...

+ curl -sLf -o /dev/null 'https://deb.nodesource.com/node_6.x/dists/stretch/Release'

## Adding the NodeSource signing key to your keyring...

+ curl -s https://deb.nodesource.com/gpgkey/nodesource.gpg.key | apt-key add -
OK

## Creating apt sources list file for the NodeSource Node.js v6.x repo...

+ echo 'deb https://deb.nodesource.com/node_6.x stretch main' > /etc/apt/sources.list.d/nodesource.list
+ echo 'deb-src https://deb.nodesource.com/node_6.x stretch main' >> /etc/apt/sources.list.d/nodesource.list

## Running `apt-get update` for you...

+ apt-get update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Get:3 https://deb.nodesource.com/node_6.x stretch InRelease [4647 B]
Get:4 https://deb.nodesource.com/node_6.x stretch/main Sources [765 B]
Get:5 https://deb.nodesource.com/node_6.x stretch/main armhf Packages [1007 B]
Fetched 6419 B in 1s (3968 B/s)    
Reading package lists... Done

## Run `apt-get install nodejs` (as root) to install Node.js v6.x and npm

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libuv1
Use 'sudo apt autoremove' to remove it.
The following packages will be REMOVED:
  nodejs-legacy nodered
The following packages will be upgraded:
  nodejs
1 upgraded, 0 newly installed, 2 to remove and 61 not upgraded.
Need to get 8761 kB of archives.
After this operation, 27.9 MB of additional disk space will be used.
Get:1 https://deb.nodesource.com/node_6.x stretch/main armhf nodejs armhf 6.11.4-1nodesource1 [8761 kB]
Fetched 8761 kB in 5s (1551 kB/s) 
apt-listchanges: Can't set locale; make sure $LC_* and $LANG are correct!
apt-listchanges: Reading changelogs...
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
(Reading database ... 122683 files and directories currently installed.)
Removing nodered (0.17.4) ...
Removing nodejs-legacy (4.8.2~dfsg-1) ...
(Reading database ... 115794 files and directories currently installed.)
Preparing to unpack .../nodejs_6.11.4-1nodesource1_armhf.deb ...
Unpacking nodejs (6.11.4-1nodesource1) over (4.8.2~dfsg-1) ...
Processing triggers for mime-support (3.60) ...
Processing triggers for desktop-file-utils (0.23-1) ...
Setting up nodejs (6.11.4-1nodesource1) ...
Processing triggers for gnome-icon-theme (3.12.0-2) ...
Processing triggers for man-db (2.7.6.1-2) ...
Processing triggers for gnome-menus (3.13.3-9) ...
Node.js installation Done!
Cloning MagicMirror ...
Cloning into 'MagicMirror'...
remote: Counting objects: 8632, done.
remote: Total 8632 (delta 0), reused 0 (delta 0), pack-reused 8632
Receiving objects: 100% (8632/8632), 11.27 MiB | 1.39 MiB/s, done.
Resolving deltas: 100% (4887/4887), done.
Cloning MagicMirror Done!
Installing dependencies ...
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global jsonlint@1.6.2 should be installed with -g

> uws@0.14.5 install /home/pi/MagicMirror/node_modules/uws
> node-gyp rebuild > build_log.txt 2>&1 || exit 0

Unable to install dependencies!
```

tuddemar, 2017-10-08:
I had the same issue when trying to install Magicmirror on my freshly installer Rpi 3.

MichMich, 2017-10-08:
Which os are you trying? (It would really help if you filled in all the required information when creating an issue. Thanks!)

twalenzyk, 2017-10-08:
Sorry I forgot to mention.
I was using the lates raspbian package, which I donwloaded yesterday.
The Image is called: 2017-09-07-raspbian-stretch.img

I have Installed MagicMirror2 yesterday for a first try and it was working.
After playing around with config it refused working yesterday.
Today I wrote the raspbian image againto the SD Card and tried installing MM, what failed.

MichMich, 2017-10-08:
We got some reports from issues with Electron on "Stretch". You might want to try "Jessie"

twalenzyk, 2017-10-08:
I cannot get Jessie running on my Raspberry 3.

Do you have any rough schedule, when MM will work with Stretch?

WhitePonny, 2017-10-09:
I had the same problem.
Raspberry Pi 2 running Jessie and MagicMirror then upgrade to Stretch and it is still works but Raspberry Pi 3 with Stretch from beginning doesn't wanna run Electron.
I will try installing Jessie and MM before doing the upgrade on my Pi3.

MichMich, 2017-10-10:
Electron needs to be Fixed to run on Stretch. This needs to be done by the Electron team.

closed 2017-10-10 by MichMich

twalenzyk, 2017-10-13:
Hi, 

Is the electron team aware of this issue?

BR Thorsten

martincastillo3, 2018-03-19:
Hi, I had a similar issue, I dont know if you solved it, but someone may still be struggling. I read in another post a way to solve it and it worked for me.

I installed the last version of raspbian (March 2018), followed youtube's setup videos, and couldnt get the magicmirror to work. What I then did was install raspbian without updating and upgrading anything. I inserted the sd, let it boot, configure wifi and language settings and installed MagicMirror with the automatic installation. 

It works perfectly now and havent had any issue with it or any module. Hope this helps someone with a similar problem!