MagicMirrorOrg/MagicMirror #1052: Newsfeed module crashing PI 3

Saurman, 2017-10-10:
Just bought my first Pi 3 B specifically for this project. Spent whole day banging my head trying to figure this out. Eventually i've got it up and running but i am getting occasional system reboots. The core of those problems lies in newsfeed module, when disabled system is pretty stable. 

Here is what i did:
1) common MagicMirror install
2) changed Pi locale to Ukrainian (ua_UTF-8) first via UI, then by installing packages and picking locales from blue screen (don't remember exact command)
3) added newsfeed sources to a module with 2 urls: http://www.pravda.com.ua/rus/rss/view_news/, view-source:http://news.liga.net/news/rss.xml
4) did not specify encoding

As result, system crashes either on first feed message or on second. Specifying different encodings either produce invalid characters or crash the system. Please advice how to display cyrillic symbols in a newsfeed.

Thanks!

morozgrafix, 2017-10-13:
I haven't had a chance to attempt to reproduce it, but first link seems like it doesn't validate on W3C RSS validator: https://validator.w3.org/feed/check.cgi?url=http%3A%2F%2Fwww.pravda.com.ua%2Frus%2Frss%2Fview_news%2F

Can you please try it with only second link to see if we can eliminate potentially invalid RSS?

morozgrafix, 2017-10-13:
Just tried it on a local instance (not on a Pi) and inspecting HTTP request on shows that it is encoded with `windows-1251`.  Try this setting and it may work.

```
{
	title: "Украинская правда",
	url: "http://www.pravda.com.ua/rus/rss/view_news/",
	encoding: "windows-1251"
}
```

Saurman, 2017-10-13:
Thanks @morozgrafix, i'll try check

Saurman, 2017-10-14:
@morozgrafix I can confirm that it works perfectly, thank you very much!

closed 2017-10-14 by Saurman

morozgrafix, 2017-10-16:
Glad it all worked out.