MagicMirrorOrg/MagicMirror #1056: No method to gracefully shutdown modules

shbatm, 2017-10-13:
I have developed a few modules that spawn other processes and I could not find any existing method to gracefully exit when stopping or restarting the MagicMirror.  Because of the way Electron handles SIGINT commands, it is not as simple as adding a `process.on('SIGINT', function...)` to the node_helper.  Unless you are running `server_only`, this never gets called.

I would like to propose: when exiting (e.g. SIGINT, CTRL+C, pm2 stop, or pm2 restart), check each `node_helper` for a `stop` function, and call it if it exists; similar to the `start` function.  This will allow module developers to implement a "stop" function if necessary to clean up their modules.

Working prototype here: https://github.com/shbatm/MagicMirror/commit/9935d2fccb4206c2949b43cc118aad6789303f12 

* Works when using Electron (`app.on("before-quit"`) or server-only (`process.on("SIGINT"`)
* Prevents modules stalling by force-quitting after 3 seconds if the `stop` functions are taking too long.

Please let me know your thoughts or if there is already a better way to do this.  I can start a PR if this is something that would be useful.

MichMich, 2017-10-13:
Sounds perfect! Please send a PR which includes documentation. Thanks!

MichMich, 2017-10-16:
Merged.

commit 5f13fd2d (2017-10-16): Merge pull request #1058 from shbatm/develop_1056

Gracefully Shutdown Module node_helpers on exit/SIGINT (Fixes #1056)
commit f646360a (2017-10-16): Gracefully shutdown node_helpers (Fixes #1056)

Updated documentation

Corrected Typo in Documentation

Style correction
closed 2017-10-16 by MichMich