MagicMirrorOrg/MagicMirror #1059: Very bad performance on raspberry pi

Kugelfang666, 2017-10-13:
Hi everyone,

i just set up MM on a fresh installed and updated Raspberry Pi3. unfortuately the performance of MM is very slow. Animated content looks as it runs with 10frames per second. Sometimes even the second indicator of the clock just stops and then updates a few seconds at once. The CPU seems not to be loeaded or anything and the actual GUI of the OS runs smooth. 

Any idears how to solve this?

MichMich, 2017-10-15:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-10-15 by MichMich

mikecebul, 2017-10-24:
This version did not fix the issue with electron overworking the cpu. Downgrading to 2.1.0 and electron 1.4.15 is still the current fix.

Czara, 2017-11-22:
Hello, 

I've the same Issue, if I downgrade don't work for me I have this error

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@3.10.10
3 info using node@v6.12.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.0~prestart: magicmirror@2.1.0
6 silly lifecycle magicmirror@2.1.0~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.1.0~start: magicmirror@2.1.0
8 verbose lifecycle magicmirror@2.1.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.1.0~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.1.0~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.1.0~start: Args: [ '-c', 'electron js/electron.js' ]
12 silly lifecycle magicmirror@2.1.0~start: Returned: code: 1  signal: null
13 info lifecycle magicmirror@2.1.0~start: Failed to exec start script
14 verbose stack Error: magicmirror@2.1.0 start: `electron js/electron.js`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:191:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:191:7)
14 verbose stack     at maybeClose (internal/child_process.js:920:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:230:5)
15 verbose pkgid magicmirror@2.1.0
16 verbose cwd /home/pi/MagicMirror
17 error Linux 4.9.59-v7+
18 error argv "/usr/bin/node" "/usr/bin/npm" "start"
19 error node v6.12.0
20 error npm  v3.10.10
21 error code ELIFECYCLE
22 error magicmirror@2.1.0 start: `electron js/electron.js`
22 error Exit status 1
23 error Failed at the magicmirror@2.1.0 start script 'electron js/electron.js'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the magicmirror package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     electron js/electron.js
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs magicmirror
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls magicmirror
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

Thank you