MagicMirrorOrg/MagicMirror #1062: Displaying Icons Based on Current Max Temp

nicho29, 2017-10-17:
I am new to the Magic Mirror concept and have a basic understanding of javascript.

I am interested in creating a Magic Mirror for my young kids. One of the things they always want to know is whether they can wear pants of shorts on a given day.

My thought is to reprogram the Current Weather modele to retrieve the temp_max for the current day and when it is above a certain number of degrees for it to display an icon of my choice. If the temp_max for the day is less that a set number of degrees for it to show a different icon.

I've been playing with the main js code and am attempting to mirror a few of the items and combine other parts to get this working.

Here is what I have so far.
...
		this.temperature = null;
		this.maxtemperature = null;
		this.indoorTemperature = null;
...
		var maxtemperature = document.createElement("span");
		maxtemperature.className = "bright";
		maxtemperature.innerHTML = " " + this.maxtemperature + "&deg;" + degreeLabel;
		large.appendChild(maxtemperature);
...
		this.humidity = parseFloat(data.main.humidity);
		this.temperature = this.roundValue(data.main.temp);
		this.maxtemperature = this.wearTable(data.main.temp_max)
...
	wearTable: function(temp_max) {
		if (temp_max>=70){
			return "NNE";
		} else if (temp_max<70) {
			return "NE";
		}
	},
This last part is incomplete but I was mirroring it off the deg of wind direction only it would return or display different icons that I could create and store in the appropriate folder.

Any help on this would be appreciated.

Thanks!
[Mirror.txt](https://github.com/MichMich/MagicMirror/files/1392380/Mirror.txt)

nhubbard, 2017-10-17:
I'll think about this.
Just a little note: if you want a code block, use this inside Markdown:
<pre>
```javascript
/* insert code here */
```
</pre>
It makes it more readable for both you and me :smile:

MichMich, 2017-10-17:
Nice idea! In my idea the best approach would be developing a new module for this which lustend to a current weather notification sent by the weather module. I’m not sure if the weather module currently sends this event, but that would be easy to implement.

MichMich, 2017-10-18:
To keep the Issuelist on GitHub clean, ill close this issue (since it isnt a bugreport).

If you need any support building your module, check out the forum: http://forum.magicmirror.builders or when you need help with the notification change, check out the Slack channel.

closed 2017-10-18 by MichMich