MagicMirrorOrg/MagicMirror #1066: Calendar problem

KalleHeuwes, 2017-10-22:
**Platform**: Raspberry Pi 3

**Node Version**: 6.11.0

**MagicMirror Version**: 2.1.3

**Description**: 
I configured 7 calendars; 6 of these are displayed correctly.
But one of them produces this error:

Whoops! There was an uncaught exception...
Error: dates lower than Thu Jan 01 1970 00:00:00 GMT+0100 (CET) are not supported
    at Object.toOrdinal (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:105:15)
    at Iterinfo.rebuild (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:1409:35)
    at RRule._iter (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:978:10)
    at RRule.between (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:794:21)
    at /home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:135:24
    at Request._callback (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:11:5)
    at Request.self.callback (/home/pi/MagicMirror/node_modules/request/request.js:188:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:194:7)
    at Request.<anonymous> (/home/pi/MagicMirror/node_modules/request/request.js:1171:10)
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?

**Steps to Reproduce**: 
The problem is this calendar (url has been cut off)
					{	// Kalle
						symbol: 'bicycle',
						url: 'webcal://calendar.google.com/calendar/ical/4q6vd8hd362o3np1die1nkubes%40group.calendar.google.com/private-.../basic.ics'
					}

**Expected Results**: The calendar entries

**Actual Results**: No entries at all; just the error message metioned above

**Configuration**: See "steps to reproduce"

nhubbard, 2017-11-18:
Are any of the dates in your calendar before January 1, 1970? This is a hardware limitation that we cannot bypass in any way.

KalleHeuwes, 2017-11-23:
Hi, Nicholas !

Yes there are (birth ) dates before 19700101 as some of my friends are very
old.
Maybe I should get rid of my old friends ;-)

Kind regards
Kalle

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virenfrei.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

nhubbard, 2017-12-07:
@HalleHeuwes You don't have to get rid of your old friends :) My dad was born before 1970; I changed his birthday to the same date in 1970 and calendar parsing issues disappeared. Could you try that?

KalleHeuwes, 2017-12-08:
Hi, Nicholas!

Thank you for your support.
I don't want fake the birth dates, just to make them fit into mm.
I created a different calendar in google holding all my dates.
The main calendar, which includes the birth dates isn't included in mm anymore.
This works for me.

Kind regards
Kalle

MichMich, 2017-12-08:
Good workaround. Closing this for now. Feel free to reopen.

closed 2017-12-08 by MichMich

bertwest, 2018-01-10:
Hi all,
I stumbled upon this issue as well. I do not want to have a separate calendar as @KalleHeuwes suggests because I want all this data to be synced automatically and so far this works out nicely family-wide.
And I also don't like the idea to fake the dates.
Would be cool if this topic could fixed some time 
Bregards,
Bert

CharonBeber, 2018-02-23:
I also had delete all the old dates before 1970. Now works like charm. Especially the birth dates and yearly reminders created a problem.