MagicMirrorOrg/MagicMirror #1067: Calendar List is not limited to maximumEntries

wbleek, 2017-10-22:
I have updated to the lates version on master. Since then, the problem occurs.

**Platform**: RaspPI 3, Jesse
**Node Version**: 6.11.3
**MagicMirror Version**: v2

**Description**: I have configures multiple calendars. I have set the config value maximumEntries explicitly to 10. I get a list of more then 30 entries display since the update.

oss001, 2017-10-23:
I am having the same issue.  maximumEntries is having no effect.

morozgrafix, 2017-10-23:
I just tested it and it seems that when you set `maximumEntries` for entire calendar module config you get `maximumEntries` * number of calendars. So if you have 4 calendars and set `maximumEntries: "2"` you will get total of 8 entries. Keep in mind that node server or electron needs to be restarted in order to pickup new setting from the config, as it is read during startup if I'm not mistaken. I haven't had a chance to look through the code yet. Do you mind posting your config? but please take out your personal calendar links.

FROGGS, 2017-10-23:
This fixes it for me:
```diff
diff --git a/modules/default/calendar/calendar.js b/modules/default/calendar/calendar.js
index ad3be8c..cefd73f 100644
--- a/modules/default/calendar/calendar.js
+++ b/modules/default/calendar/calendar.js
@@ -120,7 +120,7 @@ Module.register("calendar", {
        // Override dom generator.
        getDom: function () {
 
-               var events = this.createEventList();
+               var events = this.createEventList().slice(0, this.config.maximumEntries || defaults.maximumEntries);
                var wrapper = document.createElement("table");
                wrapper.className = "small";
 
```

morozgrafix, 2017-10-23:
I think this will work when you specify `maximumEntries` for entire calendar module, but may not work if you want to specify `maximumEntries` per calendar.

FROGGS, 2017-10-23:
Well, then we should only slice the events there *if* maximumEvents is set for the entire calendar.
I only know that the current behaviour does not work at all for my scenario.

morozgrafix, 2017-10-23:
I agree there is some room for improvement. Maybe `maximumEntries` for entire module should be a separate config from `maximumEntries` per calendar.

oss001, 2017-10-23:
Thanks for posting @FROGGS, that changed it to an overall max limit which was what I had expected.  I agree a separate config entry would be a great idea.  For me, I required an over all limit due to lack of display space but I see value in per calendar as well.  I greatly appreciate all your hard work!!

MichMich, 2017-10-24:
Feel free to send me a PR to incorporate this in the core. 

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-10-24 by MichMich