MagicMirrorOrg/MagicMirror #1068: News Feed shows invalid Date for certain entires

wbleek, 2017-10-22:
**Platform**: RaspPi 3, Jesse
**Node Version**: 6.11.3
**MagicMirror Version**: V2

**Description**: Since the update to the newest version on master some newsfeed entries display "invalidDate".
url: "http://www.apotheke-adhoc.de/feed/topnews/"
The newsfeed provides the dates with German text but as far as I remember it did not throw an error beforehand. However, it could be that the newsfeed changed its format around the same time the Magic Mirror2 update was performed.

**Steps to Reproduce**: please use the above listed newsfeed URL

**Expected Results**: date should show as "invalid Date"

**Actual Results**: correct date should be shown

morozgrafix, 2017-10-23:
It looks like that RSS feed is using DIN 1355 date format like `<pubDate>Mo, 23 Okt 2017 14:10:00 +0200</pubDate>` which is invalid for RSS standards where pubDate must be an RFC-822 date-time. When `newsfeed.js` is attempting to convert it into a meaningful date object it chokes and produces invalid date.

```{javascript}
new Date('Mo, 23 Okt 2017 14:10:00 +0200')
=> Invalid Date
new Date('Mon, 23 Oct 2017 14:38:06 GMT')
=> Mon Oct 23 2017 07:38:06 GMT-0700 (PDT)
```

RSS validator:
https://validator.w3.org/feed/check.cgi?url=https%3A%2F%2Fwww.apotheke-adhoc.de%2Fnachrichten%2Frss.xml

MichMich, 2017-10-24:
As @morozgrafix notes: only valid rss files are supported.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-10-24 by MichMich

wbleek, 2017-10-24:
I just want so say thanks for investigating and pointing it out. I'll get in contact with the feed source.