MagicMirrorOrg/MagicMirror #1070: Events from Google Calendar showing at incorrect dates, or not showing at all

Txukie, 2017-10-24:
I have specifically created a Google Calendar to test this out and populated it with a few entries both recurring and non-recurring. None of them seem to show up in my calendar view. The same goes with other (private) Google calendars I use. Sometimes events show up but incorrectly, like several months in advance although they seem to show up on the same day, as in an event created for the 25th of March will actually show up as occurring on the 25th of October for example.

```
{
        module: "calendar",
        header: "Calendario",
        position: "top_left",
        classes: 'default everyone',
        config: {
            maximumEntries: 5,
            maximumNumberOfDays: 100,
            calendars: [
                  {
                           symbol: "exclamation-circle",
                           url: "webcal://calendar.google.com/calendar/ical/cf9euqob2a0tqe2409jffnpk2o%40group.calendar.google.com/public/basic.ics"
                   }
                  ]
              }
}
```

Just FYI, my locales are set as follows:

```
LANG=es_ES@euro
LANGUAGE=
LC_CTYPE="es_ES@euro"
LC_NUMERIC="es_ES@euro"
LC_TIME="es_ES@euro"
LC_COLLATE="es_ES@euro"
LC_MONETARY="es_ES@euro"
LC_MESSAGES="es_ES@euro"
LC_PAPER="es_ES@euro"
LC_NAME="es_ES@euro"
LC_ADDRESS="es_ES@euro"
LC_TELEPHONE="es_ES@euro"
LC_MEASUREMENT="es_ES@euro"
LC_IDENTIFICATION="es_ES@euro"
LC_ALL=
```

morozgrafix, 2017-10-24:
@Txukie it may not be a good idea to post link to your private calendar in a public forum.

Txukie, 2017-10-25:
You're totally right, let me make another account and copy this calendar accross and I will update my report accordingly.

javilandoando, 2017-11-19:
I have same issue.
I don't know why although on google calendar any birthday remind is displayed properly, using this module I get a reminder related to a birthday on a different month although day, 21st for example, is correct.

Does any body find how to correct this?

kleinmantara, 2017-11-29:
The same problem with me. I also use Google Calendar and all-day entries are displayed at the wrong time.

mattmurf, 2017-12-03:
I'm getting garbled times..  The date is showing correctly.  

This was working and broke at some point in the past couple months.

Example:
Quiz blah, Blah, Blah
Today at AMb0oo12/4/2017ut1

matthewpmangus, 2017-12-09:
I don't know if it'd work OR IF anything else MIGHT break, but it may be worth trying...

`npm update ical`?

nhubbard, 2017-12-12:
@matthewpmangus If you don't have testing hardware, it works just fine on an Intel/AMD based system. We don't use any special modules.

Txukie, 2018-01-03:
Tried the npm update ical to no avail

E3V3A, 2018-01-07:
Make sure your locale is set correctly. There is little reason to not use UTF8.
```bash
# echo $LANG
en_US.UTF-8
```

Also check you time/date settings with:
```bash
# timedatectl status
...
```

javilandoando, 2018-01-08:
time/date settings in my case are correct due to if they would be wrong, date will be displayed wrong on magic mirror.

I cannot check currently my locale setting but how does it impact on a time issue?
Does any body solve this problem?

SpoturDeal, 2018-01-24:
I have noticed that events that are in the iCal records are without an DTEND will disappear on the actual date it should be visible. Without DTEND they will work in iPhone, Android and Windows.

```
BEGIN:VEVENT
UID:7c1caf04b1671cd607c204778c1ca9d1@www.uden.nl
DTSTAMP:20180118T102124
DTSTART;VALUE=DATE:20180317
SUMMARY:Papierinzameling
DESCRIPTION:Ophaaldag: zaterdag
TRANSP:TRANSPARENT
END:VEVENT
```
But this one works just add an DTEND to the event. (it is a mistake in building the iCal file by the provider.) In this case the city of Uden Netherlands

```
BEGIN:VEVENT
UID:7c1caf04b1671cd607c204778c1ca9d1@www.uden.nl
DTSTAMP:20180118T102124
DTSTART;VALUE=DATE:20180317
DTEND;VALUE=DATE:20180318 
SUMMARY:Papierinzameling
DESCRIPTION:Ophaaldag: zaterdag
TRANSP:TRANSPARENT
END:VEVENT
```

MichMich, 2018-02-07:
Since this all has to do with the ical parser, I suggest we continue this issue discussion in: https://github.com/MichMich/MagicMirror/issues/1105

closed 2018-02-07 by MichMich