MagicMirrorOrg/MagicMirror #1071: Filter news entries

TopperDEL, 2017-10-30:
This is not a but, but a Feature request. I hope this place is still ok.

I would like to see a filter-Attribute for the newsfeed-module. There I would add a list of words an if any of These are found within the tilte of a News entry, the News entry gets ignored.

I want to hide any sport-related News on my mirror as this is of no interest for me. But there is no "normal" newsfeed in Germany without "News" about Soccer and such...

Thank you!

ToddKlindt, 2017-10-30:
I would love this too. From one of my local newspaper's RSS feed I get headlines about pictures that I can't see, or editorial stuff. It would be great to be able to filter out articles by words in the title.

morozgrafix, 2017-11-01:
Code for this feature has been merged into `develop` branch. Please test it out and let me know if this is something you had in mind. Also please let me know if you think it can be improved in any way.

Thanks.

TopperDEL, 2017-11-01:
I checked it out and tried it. Seems to work exactly as expected - thank you! I have three newsfeed-modules with filter and one without - all work seemlessly.

morozgrafix, 2017-11-02:
@MichMich I think we can close this one.

PR #1072 (2017-11-01): Added ability to set a list of prohibited words that will be filtered…
… out of newsfeed

Resolves #1071

`prohibitedWords` config parameter is an array of words. If set and case insensitive greedy match is found anywhere in the title then that newsfeed item will not be displayed. 

Readme updated with instructions.

Users should be careful on the words selection as careless setting may remove many or all items from the newsfeed. Some obvious mistakes like `space, comma, dot` etc. can be prevented programatically, but I left it out of this PR

Example:

with `prohibitedWords: ['dodgers']`

Ori [... cut]
commit 0eb1c0ce (2017-11-01): Added ability to set a list of prohibited words that will be filtered out of newsfeed

Resolves #1071

`prohibitedWords` config parameter is an array of words. If set and case insensitive greedy match is found anywhere in the title then that newsfeed item will not be displayed. Readme updated with instructions.

Users should be careful on the words selection as careless setting may remove many or all items from the newsfeed. Some obvious mistakes like `space, comma, dot` etc. can be prevented programatically, but I left it out of this PR

Example:

with `prohibitedWords: ['dodgers']`

Original [... cut]
closed 2017-11-02 by MichMich