MagicMirrorOrg/MagicMirror #1075: Possible Bug in clientonly/index.js

TobiTobiM, 2017-11-06:
First of all, thanks for the project. Makes fun setting up/building the magic mirror. 

**Platform Server:** x86_64 Docker Container based on bastilimbach/docker-magicmirror Image
**Plattform Client:** Raspberry PI 3 Raspbian Electron 1.7.9
**Node Version:** 9.0.0
**MagicMirror Version:**: Head Version (Master) both Client and Server

**Description:**: Setup looks like this. To give the small raspi some air to breathe, the server side is running on a docker host in my network. The docker host was already there so i thought why not. I made a rebuild of the image based on the docker file provided by bastilimbach/docker-magicmirror with  less changes but one is important for the possible bug. I had to change the address for the server to 0.0.0.0. Why? Because the interface inside the started docker container could have any address from docker virtual network. Port 8080 is mapped to 9000 because a lot of ports are already in use on my docker host.
If i call http://docker_host_address:9000 from any computer in my network in normal browser like chrome or firefox magic mirror shows up as expected. 
If i start magic mirror on the raspi with 

> DISPLAY=:0 node clientonly -v --address docker_host_address --port 9000

 the screen stays black no module shows up. A quick tcpdump shows that docker host is called and it loads config.js from the server. Communication is possible. Fine so why isn't it working? 
Fortunately magic mirror is open source so i could take look in the code. The bug/problem is how clientonly/index.js handles the loaded configuration from server. 
Lines of code are from the actual master branch.

In line 64 the configuration is loaded from server it uses the address and port given from command line parameter.

> getServerConfig(`http://${config.address}:${config.port}/config/`)

At this point config.address is docker_host_address and port is 9000. Which are the right values in my case. 

Then after loading address and port in line 69/70 are set to the value from loaded server config.

> config.address = config.address;
> config.port = config.port;

Now the config.address is set to 0.0.0.0 and port is set 8080. And this is the reason why the screen stays black electron tries to contact http://0.0.0.0:8080.
I know this setup is not typical out there but it could happen in docker and vm enviroments where ports/address are mapped/forwarded.
I quick and dirty workaround is to set config.address and config.port to fix values. But after an update i have to redo it again.

**Expected Results**: Magic mirror shows all configured modules

**Actual Results**: Screen stays black.

**Configuration**: 

> var config = {
>         address: "0.0.0.0", //need this to bind nodejs to all interfaces in docker container
>         port: 8080, // this is mapped to port 9000 outside container
>         ipWhitelist: [], // access from all in network/vlan
> 
>         language: "de",
>         timeFormat: 24,
>         units: "metric",
> .
> .
> .
> }

plmokn102030, 2017-11-13:
I'dont start smart mirror 
Give me advice please thanks
Raspberry pi terminal 

bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"

MichMich, 2017-11-23:
@TobiTobiM the docker integration is a community contribution, so not something I know a lot about. If you know a solution, could you send a PR?

@plmokn102030  please refer to http://forum.magicmirror.builders for support and troubleshooting.

MichMich, 2017-12-08:
Closed due to inactivity. Feel free to reopen.

closed 2017-12-08 by MichMich

soilsk, 2018-07-07:
@TobiTobiM: you save my day!! Thank you ;]