MagicMirrorOrg/MagicMirror #1082: sh run-start.sh fails

makerschwabe, 2017-11-17:
Please only submit reproducible issues. 

I tryed to set up magic mirror on a raspberry pi zero with npm v4.1.2

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

Axel-Germany, 2017-11-17:
The file npm-debug.log shows this:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@4.1.2
3 info using node@v7.7.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.3~prestart: magicmirror@2.1.3
6 silly lifecycle magicmirror@2.1.3~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.1.3~start: magicmirror@2.1.3
8 verbose lifecycle magicmirror@2.1.3~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.1.3~start: PATH: /usr/local/lib/node_modules/npm/bin/node-gyp-bin:/home/$
10 verbose lifecycle magicmirror@2.1.3~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.1.3~start: Args: [ '-c', 'sh run-start.sh' ]
12 info lifecycle magicmirror@2.1.3~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.1.3 start: `sh run-start.sh`
13 verbose stack spawn ENOENT
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/spawn.js:33:16)
13 verbose stack     at emitTwo (events.js:106:13)
13 verbose stack     at ChildProcess.emit (events.js:194:7)
13 verbose stack     at maybeClose (internal/child_process.js:899:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
14 verbose pkgid magicmirror@2.1.3
15 verbose cwd /home/pi/MagicMirror
16 error Linux 4.9.35+
17 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 error node v7.7.3
19 error npm  v4.1.2
20 error file sh
21 error code ELIFECYCLE
22 error errno ENOENT
23 error syscall spawn
24 error magicmirror@2.1.3 start: `sh run-start.sh`
24 error spawn ENOENT
25 error Failed at the magicmirror@2.1.3 start script 'sh run-start.sh'.
25 error Make sure you have the latest version of node.js and npm installed.
25 error If you do, this is most likely a problem with the magicmirror package,
25 error not with npm itself.
25 error Tell the author that this fails on your system:
25 error     sh run-start.sh
25 error You can get information on how to open an issue for this project with:
25 error     npm bugs magicmirror
25 error Or if that isn't available, you can get their info via:
25 error     npm owner ls magicmirror
25 error There is likely additional logging output above.
26 verbose exit [ 1, true ]

Axel-Germany, 2017-11-17:
I tryed to run Magic Mirror on a Raspberry Pi Zero W

pi@weberpi:~ $ git clone https://github.com/MichMich/MagicMirror
fatal: destination path 'MagicMirror' already exists and is not an empty directory.
pi@weberpi:~ $ cd ~/MagicMirror
pi@weberpi:~/MagicMirror $ 
pi@weberpi:~/MagicMirror $ npm install && npm start
ERROR: npm is known not to run on Node.js v4.2.4
Node.js 4 is supported but the specific version you're running has
a bug known to break npm. Please update to at least 4.7.0 to use this
version of npm. You can find the latest release of Node.js at https://nodejs.org/
pi@weberpi:~/MagicMirror $ 
pi@weberpi:~/MagicMirror $ sudo curl -sL https://deb.nodesource.com/setup_4.x | sudo bash -

## Installing the NodeSource Node.js v4.x LTS Argon repo...

## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js v4 and later.

pi@weberpi:~/MagicMirror $ npm install && npm start
Killed         ....] \ extract:ansi-styles: sill gunzTarPerm modified mode [ 'l
pi@weberpi:~/MagicMirror $ npm start

> magicmirror@2.1.3 start /home/pi/MagicMirror
> sh run-start.sh

run-start.sh: 4: run-start.sh: electron: not found

npm ERR! Linux 4.9.35+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v7.7.3
npm ERR! npm  v4.1.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! magicmirror@2.1.3 start: `sh run-start.sh`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.3 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
pi@weberpi:~/MagicMirror $ node -v~

MichMich, 2017-11-17:
Thanks @Axel-Germany for solving this issue. Closing it for now.

closed 2017-11-17 by MichMich