MagicMirrorOrg/MagicMirror #1083: Calculation issue in calendar module

Trickx, 2017-11-17:
**Platform**: raspbian on Raspberry Pi 3
**Node Version**: node -v -> v6.12.0

**MagicMirror Version**: v2

**Description**: Today we have Friday the 17th of November.
There is a full-day (no time slot) entry "XXX Geburtstag" in my iCloud calendar for Monday 20th.
The entry is shown as in 2 days, but it's in three.
Entries for the same day e.g. "Tennis", but with a defined time slot e.g. 4-5 pm, are shown correctly (in 3 days).

[image]
 
**Expected Results**: Tills Geburtstag to be shown as "in 3 days".

**Actual Results**: Tills Geburtstag is shown as "in 2 days".

**Configuration**: 
        module: "calendar",
                        header: "Familienkalender",
                        position: "top_left",
                        config: {
                                timeFormat: "absolute",
                                urgency: "3",
                                calendars: [
                                        {
                                                symbol: "calendar-check-o ",
                                                url: "webcal://p31-calendars.icloud.com/URL......."
                                        }
                                ]
                        },

Kugelfang666, 2017-11-18:
thanks for bringing this up, i have the same issue!

MichMich, 2017-11-23:
Could you check if this is incorrect both in the morning and in the evening? And if not, what si the date and time of the event in your agenda?

Kugelfang666, 2017-11-25:
once I saw that the problem was only occurring after noontime. I’lol check in a few days to veryfy. In my case all the events are full day events as I’m using a set made calendar only comprising birthdays

MichMich, 2017-11-26:
My guess is that moment.js only counts a day when it's 24 hour (or more). 23,5 hours will not count as a day.

Kugelfang666, 2017-11-30:
could be, but why do whole day- appointments vanish form the calendar on the actual day? Two weeks ago i put this post on the MM forum: 

> i have the following problem:
> I set up my MM with the default calendar module wich is connected to my icloud calendar. So far the connecton works nice an all the events are showing up. However, the indicated time seems to be off. Specifically i have a calendar which only hold birthdays (each fullday events). Lets assume that today is the 11th and a friends birthday is on the 15th. My MM would tell me that the birthday is in 3 days which seems odd. Moreover, on the 15th the event would dissapear entirely. It would be great if someone could help me to fix this.
> 
> I already checked and my raspberry is set to the correct time zone just to rule out the obvious :-)

MichMich, 2017-12-04:
Please run the latest version, and downgrade electron using the following command:

    npm install electron@1.4.15

Kugelfang666, 2017-12-21:
thanks for the input, I'll give it a try. stupd question however, how do i update my MM to the latest version without having to reinstalla l the plugins?

MichMich, 2017-12-28:
https://github.com/MichMich/MagicMirror#updating-your-magicmirror

Your modules folder and config won't get touched, so don't worry.

Kugelfang666, 2017-12-29:
thanks for the help. Just updated to the most recent version and gave it a shot. Problem still exists: Full-Day events immediately disappear on the actual date of occurance whereas i would expect them to be displayed throughout the day.

Kugelfang666, 2018-01-12:
So i did some throughout testing, apparently i found the problem! Originally filled the problematic calendar using the apple icloud addon for outlook. The other day i also added an item via my iPhone. I realized that only the full-day items added via outlook exhibited this wired problem with dissappearing at 0:00. I coudl fix all dates by opening them on my iphone and resaving them again. Apparently there must be something wrong with the icloud addon from apple messing up the ical files or so. 
Anyway, the calendar plugin is now working as intedned!

MichMich, 2018-01-17:
Can I close this issue?

Kugelfang666, 2018-01-17:
From my perspective yes as my problem is solved

closed 2018-01-20 by MichMich