MagicMirrorOrg/MagicMirror #1084: Possible EventEmitter issues with (split) weather modules

nhubbard, 2017-11-18:
**Platform**: Raspberry Pi 3 Model B

**Node Version**: v6.11.4

**MagicMirror Version**: 2.1.3

**Description**: Eventually, I have started seeing these issues with my Raspberry Pi 3, and this is non-reproducible. I'm getting error messages like this from PM2:
```
mm-0 (err): (node:1176) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 pipe listeners added. Use emitter.setMaxListeners() to increase limit.
```

**Expected Results**: Weather modules appear together.

**Actual Results**: There is a blank, black space with the weather module.

**Configuration**:
```javascript
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */
var config = {
	port: 8080,
	language: "en",
	timeFormat: 12,
	units: "imperial",
	modules: [
		{
			module: "clock",
			position: "top_left",
			timeFormat: 12,
			showPeriod: false,
			clockBold: true
		},
		{
			module: "calendar",
			header: "Upcoming Events",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "users",
						url: "{{removed}}"
					}
				]
			}
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "Sterling Heights, MI",
				appid: "{{removed}}"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			config: {
				location: "Sterling Heights, MI",
				appid: "{{removed}}"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		}
	]
};
/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
```

MichMich, 2017-11-23:
EventEmitter is not something from the MM core. I suspect this has something to do with the Node version or Electron version. Could you try with another node version?

nhubbard, 2017-11-25:
Sure.

MichMich, 2017-12-08:
Closed due to inactivity. Feel free to reopen.

closed 2017-12-08 by MichMich