MagicMirrorOrg/MagicMirror #1093: Magic Mirror server

caldecy, 2017-11-26:
MagicMirror Version: Javascript version (v2) running on Raspberry Pi 3.

Description: I keep getting the following error message when running my webcam server with the Magic Mirror app.

**Error: listen EADDRINUSE 127.0.0.1:8080
    at Object.exports._errnoException (util.js:1050:11)
    at exports._exceptionWithHostPort (util.js:1073:20)
    at Server.setupListenHandle [as _listen2] (net.js:1263:14)
    at listenInCluster (net.js:1304:12)
    at doListen (net.js:1428:7)
    at GetAddrInfoReqWrap.asyncCallback [as callback] (dns.js:62:16)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:78:10)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '127.0.0.1',
  port: 8080 }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues**

Steps to Reproduce: I have installed the webcam server software "Motion" and got it running on server 8080. I understand that this is the same server that Magic Mirror app will run on, hence the error occurred. To reproduce this error, I ran _$ sudo service motion start_ first before _npm start_ in the Magic Mirror directory. 

What I hope to understand is whether do I need to change the server number on the Magic  Mirror app (guess not because it takes priority) or that on the Webcam Motion config file? And to what other server number do I change it to? I'm pretty new to this Raspberry Pi coding interface so would greatly appreciate it if your answers are more simple to understand. 

Thanks in advance!

MichMich, 2017-11-26:
Since this is a third party module issue, please refer to http://forum.magicmirror.builders for support and troubleshooting.

Quick pointer in the right direction: configure MagicMirror to use a different post, ie: 8081.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

MichMich, 2017-11-26:
https://github.com/MichMich/MagicMirror#configuration

closed 2017-11-26 by MichMich