MagicMirrorOrg/MagicMirror #1094: Issue with displaying rain in weather_forecast module

caldecy, 2017-11-27:
**MagicMirror Version**: 2.1.3

**Description**: I cannot get Magic Mirror to display the rain probability for the day in the current_weather display. I changed the _showRainAmount: true_ in the weatherforecast.js but the rain amount does not show in the display.

**Configuration**: Module.register("weatherforecast",{

	// Default module config.
	defaults: {
		location: false,
		locationID: true,
		appid: "",
		units: config.units,
		maxNumberOfDays: 3,
		**showRainAmount: true,**
		updateInterval: 10 * 60 * 1000, // every 10 minutes
		animationSpeed: 1000,
		timeFormat: config.timeFormat,
		lang: config.language,
		fade: true,
		fadePoint: 0.1, 
		colored: false,
		scale: false,

Also, when I set the maximum number of days to 3, it still displays the weather forecast for the entire week. I just want to make the forecast column look less cluttered.

MichMich, 2017-12-04:
It might have something to do with the "recent" api changes of OpenWeatherMap for new users. Please check the forum.

nhubbard, 2017-12-07:
@caldecy You have to make the changes in *your* config file (config/config.js), not in the module defaults config. Those are defaults so that those options work when they aren't changed by the user.

MichMich, 2017-12-08:
Ah, indeed, good catch @nhubbard.

That should solve the issue. Closing it for now. Feel free to reopen.

closed 2017-12-08 by MichMich

caldecy, 2017-12-09:
Thanks @nhubbard for the advice, however even after changing the config file (config/config.js) and inputting the relevant fields, I still see the weather forecast for 5 days. 

My script is as such:

```
{
			module: "weatherforecast",
			position: "top_right",	// This can be any of the regions.
			config: {
				// See 'Configuration options' for more information.
				location: "Singapore",
				locationID: "1880252", //Location ID from http://openweathermap.org/help/city_list.txt
				appid: "a5d19d74b0e099ff98af8234c468ae8d", 
				maxNumberOfDays: 3,
				showRainAmount: true
				}
		},
```

OverAss, 2019-08-30:
Stumbled upon the same problem. Neither maxNumberOfDays or showRainAmount affects my MM when changed. Here's my config:

config: {
    location: "Trondheim",
    locationID: "3133881",
    appid: "MyAppIDHere",
    maxNumberOfDays: 10,
    showRainAmount: "true"
 }

fewieden, 2019-09-11:
@OverAss I suggest to use the new `weather` module instead of the weather forecast module. The newer api keys return a different structure of data, which the new module is handling already. Also you have more options of weather providers, which could be more accurate for your location.

https://github.com/MichMich/MagicMirror/tree/master/modules/default/weather