MagicMirrorOrg/MagicMirror #1097: Help! creation of Icons

Basanth-kumar-Uppalancha, 2017-11-28:
Hi all! Here i am a beginner to this smartMirror things…just tested some modules which i liked the most…but i want to create a clean UI …just like any other os looks…i want few modules to be displaced after booting of my mirror…then there should be some icons for my modules on the initial screen …upon clicking those icon their respective modules should start their function…and after use i should b able to close them…how can i achieve this??

MichMich, 2017-11-28:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-11-28 by MichMich

Basanth-kumar-Uppalancha, 2017-11-28:
Thanks! for your reply...same thing posted in modules utilities category of support forum, but havent got any valuable answer ...so i thought better to confirm that from you, if you can suggest anything better ....than that will be going to be a base for my build....any alternatives for this? ?