MagicMirrorOrg/MagicMirror #1100: Using Roboto Condensed & subset latin-ext

ColonelMoutarde, 2017-12-04:
Hi,
By default using "Using Roboto Condensed" & "subset latin-ext" for improvment specials chars:
é, à, ù ... Using in french

MichMich, 2017-12-04:
Those chars should work with the current font. Are you experiencing issues?

ColonelMoutarde, 2017-12-04:
Without using Roboto Condensed" & "subset latin-ext
look météo...
[image]

if i add in custom.css
@import url('https://fonts.googleapis.com/css?family=Roboto+Condensed&subset=latin-ext') ; 

[image]

MichMich, 2017-12-04:
Ah, that's weird indeed. 

I think we need to mention this in the nam package of the font:
https://github.com/choffmeister/roboto-fontface-bower/issues/44

ColonelMoutarde, 2017-12-04:
the diffrence is subset=latin-ext

MichMich, 2017-12-08:
@ColonelMoutarde is it possible to set that with the NPM import?

ColonelMoutarde, 2017-12-08:
How to do this with NPM, i'm newbe.

nhubbard, 2017-12-12:
@MichMich This is difficult. The package we use doesn't have latin-ext in it. It looks like Google patches the font and hosts it instead. The solution may be using the Google Fonts version for now.

nhubbard, 2017-12-14:
Either that, or we could download the fonts and make our own package.

MichMich, 2017-12-15:
@nhubbard Looking forward to your PR! ;)

MichMich, 2017-12-30:
@ColonelMoutarde: So, a bit of a slow response, but non the less ...

I wanted to add the correct front, but before I did I wanted to see the issue with my own eyes. So I configured a module with your exact French text, and it appears to show without any issues:

[image]

Is there anything I'm missing here?

ColonelMoutarde, 2018-01-02:
For you, i add my config file
[config-Copie.js.txt](https://github.com/MichMich/MagicMirror/files/1597618/config-Copie.js.txt)

For write an accent you have 2 possibility. 
e+` and è 
first cas it's 2 chars and second cas is 1 char

ColonelMoutarde, 2018-01-02:
[config-Copie.js.txt](https://github.com/MichMich/MagicMirror/files/1597622/config-Copie.js.txt)

MichMich, 2018-01-02:
If I look at your config file the characters are scrambled too. It might be that your config file is encoded wrong. It the file stored as an utf-8 file?

ColonelMoutarde, 2018-01-02:
the file is in utf-8 without BOM. That's what notepad++ says.

MichMich, 2018-01-02:
Maybe it's the windows environment that causes the issue. Is it possible for you to test your setup in an Raspbian environment?

MichMich, 2018-01-26:
Closed due to inactivity. Feel free to reopen.

Knurlnheim, 2018-03-04:
Hello !

I have the same issue than @ColonelMoutarde, but I am on lastest Raspbian.
What can I do help investigate the issue ?

MichMich, 2018-03-06:
@Knurlnheim Did you check the encoding of your config file? What system are you on?

Knurlnheim, 2018-03-06:
System is a Raspian Strech Lite on a Raspberry Pi 2.
config.js is in UTF8.

ColonelMoutarde, 2018-03-15:
I have reinstall MagicMirror in fresh install of Raspbian Stretch on my RPI2. 
All is OK, no problems

MichMich, 2018-03-15:
@Knurlnheim Switch to the full version of Raspbian. Not the Lite version.

closed 2018-03-15 by MichMich