MagicMirrorOrg/MagicMirror #1103: Error when I run npm start on fresh raspbian stretch install

abacon118, 2017-12-13:
When I run npm start on a fresh install on raspbian stretch install i get the following in the debug file.

0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@3.10.10
3 info using node@v4.2.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.3~prestart: magicmirror@2.1.3
6 silly lifecycle magicmirror@2.1.3~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.1.3~start: magicmirror@2.1.3
8 verbose lifecycle magicmirror@2.1.3~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.1.3~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.1.3~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.1.3~start: Args: [ '-c', 'sh run-start.sh' ]
12 silly lifecycle magicmirror@2.1.3~start: Returned: code: 1  signal: null
13 info lifecycle magicmirror@2.1.3~start: Failed to exec start script
14 verbose stack Error: magicmirror@2.1.3 start: `sh run-start.sh`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
14 verbose stack     at emitTwo (events.js:87:13)
14 verbose stack     at EventEmitter.emit (events.js:172:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:87:13)
14 verbose stack     at ChildProcess.emit (events.js:172:7)
14 verbose stack     at maybeClose (internal/child_process.js:818:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
15 verbose pkgid magicmirror@2.1.3
16 verbose cwd /home/pi/MagicMirror
17 error Linux 4.9.59-v7+
18 error argv "/usr/local/bin/node" "/usr/bin/npm" "start"
19 error node v4.2.1
20 error npm  v3.10.10
21 error code ELIFECYCLE
22 error magicmirror@2.1.3 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.1.3 start script 'sh run-start.sh'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the magicmirror package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     sh run-start.sh
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs magicmirror
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls magicmirror
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

mfread, 2017-12-28:
I did a manual install. I got a warning during installation onto my RPi2B with a new install of Raspbian Stretch.

npm WARN deprecated coffee-script@1.10.0: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global jsonlint@1.0.2 should be installed with -g

The installation continued, and after some 5 mins, finished.

I'm a Linux & RPi newbie so I've no idea if I should now run 'npm install coffeescript' or not.

'npm start' worked and up popped the default mirror.

pravine, 2017-12-29:
Got same issue with a fresh installation for Rasbian Strech Lite. Any workaround on this issue ?

MichMich, 2017-12-30:
Lite is not supported. Please state all the required version numbers to allow us to find an solution. Thanks!

abacon118, 2017-12-31:
Personally, I am using the full Raspbian Stretch. When I installed I had a fresh install of Stretch.

Thank you,

Andrew Bowman

From: Michael Teeuw
Sent: Saturday, December 30, 2017 3:11 PM
To: MichMich/MagicMirror
Cc: abacon118; Author
Subject: Re: [MichMich/MagicMirror] Error when I run npm start on freshraspbian stretch install (#1103)

Lite is not supported. Please state all the required version numbers to allow us to find an solution. Thanks!

mfread, 2017-12-31:
I am using the full Raspbian Stretch install too. The mirror is working
fine. I was just curious about the Coffee script warning and thought the
developer should know.

MichMich, 2017-12-31:
You can ignore the coffee script warning. So considering the mirror works. Can I close this issue?

mfread, 2017-12-31:
that's a yes from me.

abacon118, 2017-12-31:
No. When i have a chance i will send you a picture of the error message.

Thank you,

Andrew Bowman

abacon118, 2018-01-02:
When I run the script to start it, this is what comes out in the debug.log

0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@2.14.7
3 info using node@v4.2.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info prestart magicmirror@2.1.3
6 info start magicmirror@2.1.3
7 verbose unsafe-perm in lifecycle true
8 info magicmirror@2.1.3 Failed to exec start script
9 verbose stack Error: magicmirror@2.1.3 start: `sh run-start.sh`
9 verbose stack Exit status 1
9 verbose stack     at EventEmitter.<anonymous>
(/usr/local/lib/node_modules/npm/lib/utils/lifecycle.js:214:16)
9 verbose stack     at emitTwo (events.js:87:13)
9 verbose stack     at EventEmitter.emit (events.js:172:7)
9 verbose stack     at ChildProcess.<anonymous>
(/usr/local/lib/node_modules/npm/lib/utils/spawn.js:24:14)
9 verbose stack     at emitTwo (events.js:87:13)
9 verbose stack     at ChildProcess.emit (events.js:172:7)
9 verbose stack     at maybeClose (internal/child_process.js:818:16)
9 verbose stack     at Process.ChildProcess._handle.onexit
(internal/child_process.js:211:5)
10 verbose pkgid magicmirror@2.1.3
11 verbose cwd /home/pi/MagicMirror
12 error Linux 4.9.59-v7+
13 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
14 error node v4.2.1
15 error npm  v2.14.7
16 error code ELIFECYCLE
17 error magicmirror@2.1.3 start: `sh run-start.sh`
17 error Exit status 1
18 error Failed at the magicmirror@2.1.3 start script 'sh run-start.sh'.
18 error This is most likely a problem with the magicmirror package,
18 error not with npm itself.
18 error Tell the author that this fails on your system:
18 error     sh run-start.sh
18 error You can get their info via:
18 error     npm owner ls magicmirror
18 error There is likely additional logging output above.
19 verbose exit [ 1, true ]

MichMich, 2018-01-02:
Could you try reinstalling MagicMirror? We made some changes to the dependencies. Make sure you backup your config.js file (and modules).

eugenelai, 2018-01-03:
@MichMich,

I've just done a fresh install of MagicMirror and also encountered those errors above. 

`0 info it worked if it ends with ok`
`1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]`
`2 info using npm@5.5.1`
`3 info using node@v9.3.0`
`4 verbose run-script [ 'prestart', 'start', 'poststart' ]`
`5 info lifecycle magicmirror@2.2.2~prestart: magicmirror@2.2.2`
`6 info lifecycle magicmirror@2.2.2~start: magicmirror@2.2.2`
`7 verbose lifecycle magicmirror@2.2.2~start: unsafe-perm in lifecycle true`
`8 verbose lifecycle magicmirror@2.2.2~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/b$`
`9 verbose lifecycle magicmirror@2.2.2~start: CWD: /home/pi/MagicMirror`
`10 silly lifecycle magicmirror@2.2.2~start: Args: [ '-c', 'sh run-start.sh' ]`
`11 silly lifecycle magicmirror@2.2.2~start: Returned: code: 1  signal: null`
`12 info lifecycle magicmirror@2.2.2~start: Failed to exec start script`
`13 verbose stack Error: magicmirror@2.2.2 start: 'sh run-start.sh' `
`13 verbose stack Exit status 1`
`13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:280:16)`
`13 verbose stack     at EventEmitter.emit (events.js:159:13)`
`13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)`
`13 verbose stack     at ChildProcess.emit (events.js:159:13)`
`13 verbose stack     at maybeClose (internal/child_process.js:943:16)`
`13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)`
`14 verbose pkgid magicmirror@2.2.2`
`15 verbose cwd /home/pi/MagicMirror`
`16 verbose Linux 4.9.73-v7+`
`17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"`
`18 verbose node v9.3.0`
`19 verbose npm  v5.5.1`
`20 error code ELIFECYCLE`
`21 error errno 1`
`22 error magicmirror@2.2.2 start: 'sh run-start.sh' `
`22 error Exit status 1`
`23 error Failed at the magicmirror@2.2.2 start script.`
`23 error This is probably not a problem with npm. There is likely additional logging output above.`
`24 verbose exit [ 1, true ]`

**Info as follows:**
**Platform:** Raspberry Pi 3

**Node Version:** v9.3.0

**MagicMirror Version:** v2.2.2 

**Description:** Upon boot, I have errors in my log file as seen above.

**Steps to Reproduce:** Boot up RPi3

**Expected Results:** No errors

**Actual Results:** No actually degradation. Still works ok but thought you might like to know. 

**Configuration:** Config file can be seen here [https://hastebin.com/efinepepek.js](https://hastebin.com/efinepepek.js)

**Additional Notes:** Also getting a Fontconfig error:
`
0|mm       | Fontconfig warning: ignoring UTF-8: not a valid region tag
`
`
0|mm       | Fontconfig warning: ignoring UTF-8: not a valid region tag
`

E3V3A, 2018-01-06:
I just did all this yesterday on the **latest  full Raspbian Stretch** on a Rpi3, and it works fine. 
However, you must **make sure you have first successfully installed and updated**

- Node.js (v9.3.0)
- NPM (5.6.0)
- PM2 (2.9.1)

Yes, the installation script may need some tweaking for the Stretch...but it shouldn't fail. 
If the script fails try installing everything manually. Did you setup your Rpi3 according to docs here?

MichMich, 2018-01-06:
An updated installer script PR is extremely welcome! And would be rewarded with a bunch of internet-hugs!

E3V3A, 2018-01-06:
@MichMich 
Yeah, I've been working on one, but I don't have an ETA yet.
 
What has been most annoying in the installation process, was to make sure all the node stuff was already installed properly, before running anything else. And that is the more HW and OS (release) dependent aspect of it. There are simply too many contorted (and outdated) descriptions of how to do that out there.

leveneg, 2018-01-07:
I too am getting about the same error log on full stretch.

Node.js (v9.3.0)
NPM (5.6.0)
PM2 (2.9.1)

```
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.2.2~prestart: magicmirror@2.2.2
6 info lifecycle magicmirror@2.2.2~start: magicmirror@2.2.2
7 verbose lifecycle magicmirror@2.2.2~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.2.2~start: PATH: /home/pi/.nvm/versions/node/v9.3.0/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/home/pi/.nvm/versions/node/v9.3.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle magicmirror@2.2.2~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.2.2~start: Args: [ '-c', 'sh run-start.sh' ]
11 silly lifecycle magicmirror@2.2.2~start: Returned: code: 1  signal: null
12 info lifecycle magicmirror@2.2.2~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.2.2 start: `sh run-start.sh`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/home/pi/.nvm/versions/node/v9.3.0/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:285:16)
13 verbose stack     at EventEmitter.emit (events.js:159:13)
13 verbose stack     at ChildProcess.<anonymous> (/home/pi/.nvm/versions/node/v9.3.0/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:159:13)
13 verbose stack     at maybeClose (internal/child_process.js:943:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
14 verbose pkgid magicmirror@2.2.2
15 verbose cwd /home/pi/MagicMirror
16 verbose Linux 4.9.59-v7+
17 verbose argv "/home/pi/.nvm/versions/node/v9.3.0/bin/node" "/home/pi/.nvm/versions/node/v9.3.0/bin/npm" "start"
18 verbose node v9.3.0
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error magicmirror@2.2.2 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.2.2 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

E3V3A, 2018-01-09:
@leveneg Without seeing the exact command you did before getting this error log, its prety much useless. Also describe and post the exact commands you used for installing/updating.

leveneg, 2018-01-09:
@E3V3A my apologies- I thought that had been included. I was running `DISPLAY=:0 npm start`, as specified in the install output.

Initial installation was done on a fresh stretch install via install script.

After that didn't work I reflashed stretch, installed latest node / npm, then installed once again via the installation script with the same result.

E3V3A, 2018-01-19:
Just FYI, I didn't use the full installation script. I basically went through it manually and skipping things that I had already changed or seemed irrelevant. That's why I can't be of much help here. Especially I recall skipping the part changing the bootup splash screen and themes. These are not ncessary, and seemed suspicious. Instead I just replaced the spalsh image manually later, once MM was working.

Mabu2, 2018-01-28:
Ran into similar problems people are describing here.  Clean install, MM won't run.  Tried re-installing over and over, no results.  Is there some incompatibility with newer pkg versions?

Mabu2, 2018-01-28:
Running into similar problems... here's another person with the same issue:

https://www.reddit.com/r/MagicMirror/comments/7ots7j/first_setup_tried_following_a_guide_on_how_to/?st=jcz3d960&sh=4fd69850

Mabu2, 2018-01-28:
Here's my error log:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v6.12.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.2.2~prestart: magicmirror@2.2.2
6 info lifecycle magicmirror@2.2.2~start: magicmirror@2.2.2
7 verbose lifecycle magicmirror@2.2.2~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.2.2~start: PATH: /usr/lib/node_modules/npm/nod
e_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr
/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/g
ames
9 verbose lifecycle magicmirror@2.2.2~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.2.2~start: Args: [ '-c', 'sh run-start.sh' ]
11 silly lifecycle magicmirror@2.2.2~start: Returned: code: 1  signal: null
12 info lifecycle magicmirror@2.2.2~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.2.2 start: `sh run-start.sh`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node
_modules/npm-lifecycle/index.js:285:16)
13 verbose stack     at emitTwo (events.js:106:13)
13 verbose stack     at EventEmitter.emit (events.js:191:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node
_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:106:13)
13 verbose stack     at ChildProcess.emit (events.js:191:7)
13 verbose stack     at maybeClose (internal/child_process.js:920:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:230:5)
14 verbose pkgid magicmirror@2.2.2
15 verbose cwd /home/pi/MagicMirror
16 verbose Linux 4.9.59-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v6.12.3
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error magicmirror@2.2.2 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.2.2 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

Mabu2, 2018-01-28:
I think one reason why people seem to be having issues is, whether or not they're booting into X or the CLI.

I finally tracked down the problem I was having to trying to start MM without having X running first.   This needs to be made clearer.

nhubbard, 2018-01-30:
@MichMich For now, we should tag this issue `cantfix` because it does not come from a problem with our codebase, instead coming from Stretch.

leveneg, 2018-01-30:
I got back around to this the other day, and a manual installation on a fresh stretch stretch install _did_ do the job. I was also able to confirm that Mabu2's claim that MM needs to be started with X running first.

At a minimum, it would be nice if README were updated to reflect that the installer problematic on stretch, and that X does in fact need to be running before starting MM.

Mabu2, 2018-01-30:
Perhaps the start up sequence should check to make sure X is running, and if not throw a specific error?

Another thing is when you're installing the OS, there is the option to boot to CLI or GUI.  If someone installs the boot to CLI option, they may never know why it won't work.  The error messages are ambiguous.

MichMich, 2018-01-31:
I agree, we should add some info to the README and add a check to the nam run start method. If anyone would be willing to send a PR, I would be very grateful. :)

E3V3A, 2018-02-13:
Hmm, yeah, IDK why people would try to install MM on a non-gui (no-X) machine, unless they start out from scratch trying to do serveronly install. Two other things:

1. After you do `git clone ...`, the `run-start.sh` script is not executable:  
```$ ls -al run-start.sh
-rw-r--r-- 1 pi pi 142 Jan  6 01:58 run-start.sh
```
2. There is no shebang in the script and `electron` is not in the path, unless provided a new PATH before execution. 
3. DISPLAY of course is not well defined if not running in GUI mode.
  
```bash
# cat run-start.sh
if [ -z "$DISPLAY" ]; then #If not set DISPLAY is SSH remote or tty
        export DISPLAY=:0 # Set by default display
fi
electron js/electron.js $1
```

well, i guess that was 3.

MichMich, 2018-02-13:
Please send a PR.

E3V3A, 2018-02-13:
Hi @MichMich ,
I'm sure you address this to anyone, but this need to be tested by the people having the issues, before posting a PR. First, I'd suggest them to do this: 

```bash
#!/bin/bash
if [ -z "$DISPLAY" ]; then #If not set DISPLAY is SSH remote or tty
        export DISPLAY=:0 # Set by default display
fi
electron js/electron.js $1
```
then do:
```bash
$ chmod 755 ~/MagicMirror/run-start.sh
```

now, try to re-run the installation.

d-Rickyy-b, 2018-02-16:
I got the same issue on Debian Jessie and Stretch. I tried to install the MagicMirror application as a new user `magicmirror` into the directory `/opt/MagicMirror/`, which didn't work.

Even after installing it ino the `/home/magicmirror` directory it's not possible to run the mirror.

Only after installing it into the ´/home/pi/` directory with user `pi` the installation succeeded and I am able to run the mirror. Can you please look into it? I used the installation script you provide in the readme file.

saburishindaria, 2018-02-19:
We tried install npm using npm start command and although it launches the application, the new screen is just a blank black page.

E3V3A, 2018-02-22:
@d-Rickyy-b  Yeah, MM should not be installed in root owned directories! Any directory above `/home/pi/` is considered owned by root, AFAICR. There is absolutely no good reason, I can think of, to run MM as root. 

But, this issue is important for people on other machines than RPi's, where default user has a different name and directories have different root permissions.

d-Rickyy-b, 2018-02-22:
@E3V3A That's correct, especially on other machines, or even on the pi where the user goes with a different username (may it be for security reasons or whatsoever), it's not possible to use MM. I would highly appreciate a fix :)

MichMich, 2018-02-22:
Just curious: did any of you try `sudo npm start`?

d-Rickyy-b, 2018-02-22:
@MichMich Yes, and it produces the same outcome for me. If needed I can provide you with the log and command line output, but It's basically the same as in the initial bug report.

E3V3A, 2018-03-05:
To prevent some of the installation "problems". It might help to merge the README file from the developer branch to the very outdated current master README. I would guess that many new MM users don't even know about branches. 

And for all of the rest of you, who end up here with problems:
* please don't forget to tell us what exact commands you ran **before** you had errors. 
* please use the triple back-ticks (```) to enclose your copy/paste errors or code.

MichMich, 2018-03-05:
April 1st the next master will be released. So if you can just wait a little bit longer ... ;)

d-Rickyy-b, 2018-03-05:
@E3V3A If you're interested in the steps I did before the issue happened:

```
$ sudo adduser magicmirror
$ sudo usermod -G sudo magicmirror
$ sudo su magicmirror
$ mkdir /opt/magicmirror; chown magicmirror: /opt/magicmirror; chmod 755 /opt/magicmirror
$ cd /opt/magicmirror
$ git clone https://github.com/MichMich/MagicMirror /opt/magicmirror
$ npm install && npm start
```
Even when I then tried `sudo npm start` it didn't work.
The error message and log message was basically the same as the issue creator posted in his report.

If you need further information, don't hesitate to ask.

PS: I am installing it on my RasPi 3, but would love to use another account (e.g. magicmirror) for it.

MichMich, 2018-03-05:
I think we should just add to the readme that MagicMirror should be installed with the default user (Pi).

d-Rickyy-b, 2018-03-05:
That would be a workaround ofc. May I ask what's the issue with installing it with another user? I mean it shouldn't be different than the default user, should it? Also do other systems need to create a user "pi" when they want to use MM?

MichMich, 2018-03-05:
It works fine on my Mac on my regular user account. I don’t see any reason why it shouldn’t work on an other account on a Raspberry Pi as well. 

That being said: I got a few reports like yours. And since it’s an edge case, it doesn’t really make sense to spent a lot of time investigating it (since it’s probably OS related, not mm related). My guess is it has something to do with the user rights of the node executables which electron uses.

d-Rickyy-b, 2018-03-05:
I'll try to investigate further. Will report my findings back (if I find any).

E3V3A, 2018-03-05:
@d-Rickyy-b  IDK why you seem to insist running this as root. Please explain, because earlier you mentioned security concerns with using a default username?? To me running as root seem far more insecure and in addition, **the root environment is totally different** from any other users, including missing environment variables.

d-Rickyy-b, 2018-03-05:
@E3V3A I don't know where you see that I want to run it as root?! I want to run it as no login and no shell user (e.g. "magicmirror"). If you are asking because I added magicmirror to group sudo, that was for testing purposes.

I agree and know that it's totally insecure running it as root user. I just want another user than "pi". I don't want to use the user "pi" as a user at all - if possible. I don't want to use the root user and I never said so in any previous comment.

E3V3A, 2018-03-21:
@d-Rickyy-b Try changing the `/home/pi` in the first two files from [here](https://github.com/MichMich/MagicMirror/search?q=%2Fhome%2Fpi&type=Code&utf8=%E2%9C%93).

MichMich, 2018-03-25:
I don't think this is a real issue in the code. If you really want to make it work with an other user, feel free to fix it and send a PR so others can benefit. Fo now, my suggestion is to just use Pi. Never use Root (not saying you do). 

Closing this for now, feel free to reopen or ask for some help on the forum.

closed 2018-03-25 by MichMich