MagicMirrorOrg/MagicMirror #1106: default clock module: cannot turn seconds off

mfread, 2017-12-28:
I cannot seem to turn off the seconds display in the default clock module when I add the config item displaySeconds: "false"

snapshot of a section from my ~/MagicMirror/config/config.js file
{
  module: "clock",
  position: "top_left",
  config: {
    timeFormat: "12",
    displaySeconds: "false",
    showPeriod: 'true"
  }
},
etc

I am running Raspian Stretch on a Pi2B. I followed the installation instructions from the main page of MichMich/MagicMirror on GitHub.

yesyesuk, 2017-12-28:
Hi, 

one thing I noticed is that you have a single quote in the
following line when it should be a double quote: 

showPeriod: 'true"

It is the next parameter after the one you say doesn't work. Could be
that it fails parsing. 

Chris 

I cannot seem to turn off the seconds display in the default clock
module when I add the config item displaySeconds: "false" 
> 
> snapshot
of a section from my ~/MagicMirror/config/config.js file
> {
> module:
"clock",
> position: "top_left",
> config: {
> timeFormat: "12",
displaySeconds: "false",
> showPeriod: 'true"
> }
> },
> etc 
> 
> I am
running Raspian Stretch on a Pi2B. I followed the installation
instructions from the main page of MichMich/MagicMirror on GitHub. 
--
[2].
 

Links:
------
[1]
https://github.com/MichMich/MagicMirror/issues/1106
[2]

mfread, 2017-12-28:
Thanks yesyesuk. That was a typo - corrected now.

mfread, 2017-12-28:
Problem sorted. My fault. Cannot quote true/false values.

```
{
  module: "clock",
  position: "top_left",
  config: {
    timeFormat: "12",
    displaySeconds: false,
    showPeriod: true
  }
},
etc
```

closed 2017-12-28 by mfread