MagicMirrorOrg/MagicMirror #1107: maximumEntries not working

fmandal, 2017-12-30:
**Platform**: Raspberry Pi 2 and 3

**Node Version**: 6.12.2

**MagicMirror Version**: JavaScript version (v2).

**Description**: Setting maximumEntries in config.js on calendar module does not work. It works when setting on a per calendar basis, but not on the module.

**Expected Results**: I want the total number of calendar events to be limited to a number.

**Actual Results**: The list goes on and on...

**Configuration**: `
    {

      module: "calendar",

      header: "Kalender",

      position: "top_left",

      config: {

        timeFormat: "absolute",

        getRelative: 1,

        urgency: 0,

        maxTitleLength: 25,

        wrapEvents: true,

        fadePoint: "0.6",

        dateFormat: "ddd D/M HH.mm",

        maximumEntries: 20,

        calendars: [

          {

            symbol: "home",

            url: "webcal://xxx"

          },

          {

            symbol: "home",

            url: "webcal://xxx"

          },

          {

            symbol: "flag",

            url: "webcal://xxx"

          },

          {

            symbol: "futbol-o ",

            url: "https://www.voetbalkrant.com/soccer/calendar/team/team_509_nl.ics"

          },

          {

            symbol: "futbol-o ",

            url: "webcal://ics.fixtur.es/v2/manchester-united.ics"

          },

          {

            symbol: "trash",

            url: "webcal://xxx"

          }

        ]

      }

    },
`

fewieden, 2017-12-30:
The global maximum entries setting is currently implemented as you would do the following.

```
calendars: [
      {
        symbol: "home",
        url: "webcal://xxx"
      },
      {
        symbol: "home",
        url: "webcal://xxx"
      },
      {
        symbol: "flag",
        url: "webcal://xxx"
      },
      {
        symbol: "futbol-o ",
        url: "https://www.voetbalkrant.com/soccer/calendar/team/team_509_nl.ics"
      },
      {
        symbol: "futbol-o ",
        url: "webcal://ics.fixtur.es/v2/manchester-united.ics"
      },
      {
        symbol: "trash",
        url: "webcal://xxx"
      }
    ]
```

inversation, 2017-12-30:
I have noticed this behaviour as well. If my 'maximumEntries' is set to '10' (maximumEntries: 10), then it will display the next 10 entries per calendar. So if I have 3 calendars, I will see 3 * 10 = 30 entries.

What I would expect to see is a total of 10 entries, which would consist of entries from any calendar listed. So essentially, the module would pull a bunch of entries from every calendar (maybe this can still be defined?), put them all into a bucket, sort the combined list by time&date, and display the next 'maximumEntries' amount.

MichMich, 2017-12-30:
It's not a bug. It's more of a feature request. I agree it would be better if `maximumEntries` was a total number of maximumEntries. And then add an `maximumEntries` per calendar. Fee free to send a PR for this. :)

inversation, 2017-12-31:
Sorry if this is a real dummy question, but.... I've made the changes on my local install and got it all working. I tried to do a pull request and I have absolutely no idea what to do. Is there a github for newbies guide somewhere I can follow? I kinda assumed that pull request would just check-out the code for me to make changes and then check-in. 

Would appreciate the help so I can learn =)

inversation, 2018-01-02:
This has been addressed in the latest build 2.2.1. I think this is okay to close now.

closed 2018-01-02 by MichMich