MagicMirrorOrg/MagicMirror #1108: I have a blank black screen when I start the MagicMirror

VicoLeChips, 2017-12-30:
I configured the MagicMirror first and it worked fine.
Then I added some new modules and it worked fine too until I added one module that load me on a black screen everytime. So I decided to delete this new module because it loaded me on a black screen when I started the MagicMirror. And even after deleting it, I still have the same issue : [http://prntscr.com/hu5mol]
Then when i run the logs that what is telling me : [http://prntscr.com/hu5o3j]
Please help me !

MichMich, 2017-12-30:
You have an error somewhere in your config file. Maybe a missing or incorrect `,` or something like that.

Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-12-30 by MichMich

VicoLeChips, 2017-12-30:
OMG !
i took some time reviezing the entire config/config.js file and found the error. 
Thank you sooo much !!