MagicMirrorOrg/MagicMirror #1109: Support multiple instances in Calendar

rgstephens, 2017-12-30:
I'd like to create multiple calendars (a "Today's Calendar" with certain types of calendar items and a "Group Calendar" with upcoming events associated with a group of people, etc.).  I've added a new `instance` configuration option so you can create multiple calendar instances each have a unique instance id.  For example:

```
    {
      module: "calendar",
      header: "Today's Calendar",
      position: "top_left",
      config: {
        instance: "today",
        maximumNumberOfDays: 1,
        calendars: [
          {
            symbol: "cutlery ",
            url:
              "http://diningroomschedule.com"
          },
          {
            symbol: "male ",
            url:
              "https://mensevents.com"
          },
          {
            symbol: "female ",
            url:
              "https://womensevents.com"
          }
        ]
      }
    },
    {
      module: "calendar",
      header: "Group Calendar",
      position: "top_left",
      config: {
        instance: "bmoor",
        timeFormat: "absolute",
        getRelative: 0,
        urgency: 2,
        fadePoint: 0.7,
        dateFormat: "ddd, MMM do",
        excludedEvents: ["Breakfast", "Lunch"],
        calendars: [
          {
            symbol: "futbol-o ",
            maximumEntries: 2,
            url: "http://www.trumba.com/calendars/seattle-seahawks-schedule.ics"
          },
          {
            symbol: "futbol-o ",
            maximumEntries: 2,
            url: "http://collegefootballcalendar.net/cal/264"
          }
        ]
      }
    },

```

MichMich, 2017-12-30:
I fail to understand what the issue is. Why does it need an "instance" identifier? You can just add multiple calendar modules to your config? Could you elaborate?

rgstephens, 2017-12-30:
If I create two calendars, options for the first calendar are over-ridden by the second. For example, if my first calendar specifies `maximumNumberOfDays: 2` and the second calendar doesn't include that option the `maximumNumberOfDays` for the first calendar is not limited to the two day maximum. Using the `instance` setting, the parameters are kept separate.

I've duplicated the issue with the `excludedEvents` option.

What led me to create this option was this [discussion](https://forum.magicmirror.builders/topic/697/multiple-module-instances-overwrites-itself-values?page=1) on the forum. One of the users in this discussion states that most modules don't support multi-instance and that they share node_helper.

MichMich, 2017-12-31:
This is more of a bug than a missing feature. The calendar URL's aren't shared between multiple instances either. So any other configuration  like `maximumNumberOfDays` should remain separate as well. The best approach is to let the frontend module handle the usage of the configuration. The node helper shouldn't do much more than fetching the data. Adding an "instance" configuration to solve it is a workaround which makes it unnecessary complicated.

Making the necessary changes is a pretty far-reaching task. But I think it's better to solve this in a clean way than to use quick and dirty solutions.

If it's really an issue for your calendar. You can always fork the calendar module and create & use your own version. 

Of course, an PR which solves the issue is more than welcome. I will therefor leave this Issue open. I'll close the current PR for now. 

One minor hint if you do want to make your own fork: you can always use the module's `identifier` in stead of an "instance" configuration. the `this.identifier` is unique for every instance of a module. This removes the need for the user to configure an instance name.

rgstephens, 2017-12-31:
@MichMich Thanks for the pointers.

bolish, 2019-01-21:
Hi @rgstephens @MichMich

I'm facing the excat same issue and can't figure it out myself because of lack of skills.
Did finally one of you made the improvement?

If not, @rgstephens, could you please guide me step by step so I can't implement the right modifications to get that solved?

Regards

rgstephens, 2019-01-21:
@bolish It's been a while since I worked on this.  I can't give you step by step but here's a link to my modified [calendar.js](https://github.com/rgstephens/MagicMirror/blob/master/modules/default/calendar/calendar.js) and a sample calendar [config](https://github.com/rgstephens/magic-mirror/blob/master/moduleConfigs/calendars.json).

bolish, 2019-01-21:
and here my config.js file

> 		{
			module: "calendar",
			header: "Aujourd'hui",
			position: "top_left",
			config: {
				timeFormat: "absolute",
				calendars: [
					{
						// Anniversaires
						symbol: "birthday-cake",
						url: "https://e.ics",
						maximumNumberOfDays: 1
					},
					{
						// Calendrier Fabrice
						symbol: "calendar-check",
						url: "https://ce.ics",
						maximumNumberOfDays: 1,
					},
					{
						// Calendrier Audrey
						symbol: "calendar-check-o",
						url: "https://e",
						maximumNumberOfDays: 1,
					},
					{
						// Calendrier Férié
						symbol: "flag",
						url: "https://e",
						maximumNumberOfDays: 1,
					},
					{
						// Vacances Scolaires
						symbol: "sun",
						url: "http://e.ics",
						maximumNumberOfDays: 1,
					},
					{
						// Repas
						symbol: "utensils",
						url: "https://e.ics",
						maximumNumberOfDays: 1,	
				}
				],
			}
		},
		{
			module: "calendar",
			header: "Demain",
			position: "top_left",
			config: {
				timeFormat: "absolute",
				calendars: [
					{
						// Anniversaires
						symbol: "birthday-cake",
						url: "https://e.ics",
						maximumNumberOfDays: 62,
					},
					{
						// Calendrier Fabrice
						symbol: "calendar-check",
						url: "https://e.ics",
						maximumNumberOfDays: 62,
					},
					{
						// Calendrier Audrey
						symbol: "calendar-check-o",
						url: "https://e.ics"
					},
					{
						// Calendrier Férié
						symbol: "flag",
						url: "https://e.ics"
					},
					{
						// Vacances Scolaires
						symbol: "sun",
						url: "http://e.ics"
					}
				],
			}

By the way, into the config.js file you shared, it seems there is no need for multiple instance as only one "calendar" instance is mentionned.

Regards

bolish, 2019-01-21:
thanks @rgstephens but I can't manage to get it work...
Whatever method I use (and I tried A LOT), maximumNumberOfDays is always overwritting the previous module instance...
Really hope there could be a fix soon...

sdetweil, 2019-09-01:
the easiest way to support mu;ltiple instances of the same module is on each sendSocketNotification to add the module identifier to the structure sent. 
and in the node helper send that identifier back as part of any sendsocketNotification payload.
the module.js checks the identifier, and if not matching, ignores the notification ..

the node helper also has to save the config data sent down by identifier  (or limit the use to just api data and response)...

if multiple instances are running,  (multiple module: mod_name in config.js) any sendsocketNotification from the node_helper WILL got to ALL instances... 

see my https://github.com/sdetweil/NewBusTimes repo which supports multiple module: in config.js

commit 3b32605b (2020-06-02): Merge pull request #2036 from DarthBrento/develop

Fix #1109 - multiple calendar instances with different config
closed 2019-11-07 by stale[bot]