MagicMirrorOrg/MagicMirror #1119: Chinese abnormality

Caibiy, 2018-01-05:
[image]
config: {
	compliments: {
		anytime: [
			"Hey,fuck the world pls!"
		],
		morning: [
			"Good morning, handsome!",
			"Enjoy your day!",
			"How was your sleep?"
		],
		afternoon: [
			"Hello, beauty!",
			"You look sexy!",
			"Looking good today!"
		],
		evening: [
			"晚上好"
		]
	}
}

MichMich, 2018-01-05:
My guess is the font doesn't include the required characters. You could try changing the fonts bij adjusting your css file.

Caibiy, 2018-01-06:
No effect
[image]

MichMich, 2018-01-06:
Than it might have something todo with the config.js encoding or the encoding used on the page. It could use some experimenting. A pull-request is more than welcome.

Caibiy, 2018-01-09:
thank you..it is worked..

MichMich, 2018-01-10:
Good to hear! What did you change? For future users having the same issue ...

closed 2018-01-10 by MichMich

Caibiy, 2018-01-12:
Modified config.js encoding