MagicMirrorOrg/MagicMirror #1122: Please reccomend MMM developers to add screenshots

E3V3A, 2018-01-06:
As **MM is a fantastic project that is highly connected with aesthetics**, it is quite annoying and surprisingly how few MMM module developers actually bother to put a screenshot on their README pages. Instead one has to go search or even make a temp install to find out how it really look. Most of the time a waste of effort as it doesn't suite what I was looking for. 

@MichMich 
Would you consider to or recommend module developers to add a screenshot to their pages?
(Actually why not demand!?)

MichMich, 2018-01-06:
It’s an open framework so demanding is not an option (and not the route we want to take). Giving a recommendation is a good idea though. The best place to put this is in the developer documentation and at the top of the 3rd party module listing. Feel free to send a PR for this.

E3V3A, 2018-01-06:
@MichMich 
Just updated README as well.
Thanks for your kind support! :) 🥇

closed 2018-01-06 by E3V3A