MagicMirrorOrg/MagicMirror #1129: A better source for example/template iCal *.ics files

E3V3A, 2018-01-09:
@MichMich 

I found a website that allow you download *iCal* formatted `*.ics` files of public/national holidays. They seem better than the one supplied by default. Please have a look and let us know you think.

Browse [countries here](https://www.officeholidays.com/ics/index.php) and select the new URL with the appropriate ISO country code in the link (here Ireland):

```
https://www.officeholidays.com/ics/ics_country_code.php?iso=IE
```

E3V3A, 2018-01-12:
It's certainly not important, but it help people locate a working ics for their own country as the original site only seem to have ones available for US and UK (and possibly very few others).

SpoturDeal, 2018-01-15:
I have made a Calender.ics with all Dutch holidays (feestdagen)
free to use through the url: http://acentra.nl/feestdagen

hesseny2000, 2018-01-19:
thanks E3V3A

SpoturDeal, 2018-01-19:
@hesseny 

(Als Ramadam, Eid Al Fitr (Suikerfeest) and Eid al Adha (Offerfeest) are in the list.
I am currently working on a system that you can also add a birthday list.

MichMich, 2018-01-26:
Hi Guys, What do you want me to do with the issue? Can I close it, or does it need to stay open? Feel free to send a PR with a new default ical feed.

nhubbard, 2018-01-30:
@MichMich We can close this, b/c I don't think a PR ever materialized for this issue.

closed 2018-01-30 by MichMich