MagicMirrorOrg/MagicMirror #1130: "The Day After Tomorrow"

amcolash, 2018-01-11:
So this is an issue for me (english speaker) for the calendar module. I never refer to 2 days ahead as "The Day After Tomorrow", I would just say "in 2 days". Wanted to check with others if this is normal, or if I am not alone with this one.

It would be an easy to locally change my personal install in `MagicMirror/translations/en.json:6`, to whatever I please but updates can always change things around.

Anyone else have thoughts? I am ok with opening a PR if people are interested in the change.

jealousofyou, 2018-01-11:
Same here. I also changed the translation to „in 2 days“, also because using „the day after tomorrow“ meant that the module became excessively wide and overlapped with others.

E3V3A, 2018-01-14:
👍

MichMich, 2018-01-14:
Yeah I agree. The "in 2 days" was added because both German and Dutch have a fairly short word for it ("Overmorgen" in Dutch). In English it doesn't really make sense. 

Feel free to send a PR. We might also consider adding a feature to overwrite translations using a custom translation file (just like the custom css file). If anyone wants to add this (in a separate PR), feel free to do so! :)

amcolash, 2018-01-14:
Just made a PR: #1139. The idea of custom translations seems useful, but personally don't have a need for it.

closed 2018-01-19 by MichMich