MagicMirrorOrg/MagicMirror #1131: magicmirror installation problem

rithunrosh, 2018-01-11:
When i am installing the magic mirror program in to my fresh raspberry pi, the following error has been occured.

Installing dependencies ...
npm ERR! Linux 4.9.41-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! node v6.12.3
npm ERR! npm  v3.10.10

npm ERR! shasum check failed for /tmp/npm-7383-db2c060e/registry.npmjs.org/ajv/-/ajv-5.5.2.tgz
npm ERR! Expected: 73b5eeca3fab653e3d3f9422b341ad42205dc965
npm ERR! Actual:   59c1b801b1f0f8fc667a8ff2c9e8fefd52711461
npm ERR! From:     https://registry.npmjs.org/ajv/-/ajv-5.5.2.tgz
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
Unable to install dependencies!

MichMich, 2018-01-14:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-01-14 by MichMich

HuNanHank, 2018-03-29:
> electron@1.4.15 postinstall /home/pi/MagicMirror/node_modules/electron
> node install.js

/home/pi/MagicMirror/node_modules/electron/install.js:46
  throw err
  ^

Error: read ECONNRESET
    at _errnoException (util.js:1024:11)
    at TLSWrap.onread (net.js:615:25)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron@1.4.15 postinstall: `node install.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the electron@1.4.15 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-03-29T08_15_22_050Z-debug.log