MagicMirrorOrg/MagicMirror #1140: DOM_OBJECTS_CREATED being fired off too early

ConnorChristie, 2018-01-15:
Due to https://github.com/MichMich/MagicMirror/pull/1116 the original `DOM_OBJECTS_CREATED` event is being sent to early causing some existing modules not be able to access DOM objects.

**Affected Modules**:
- MMM-Carousel

Root cause analysis:
> Just pulled an update to the develop branch and found that this commit had broken my version of the MMM-Carousel module. The module responds to a "DOM_OBJECTS_CREATED" message to start the module transitions; however, after this update, "DOM_OBJECTS_CREATED" is sent too early; before the DOM is actually populated with the result of the module.getDom() call and every function which depends on this modules DOM being in place fails.
> 
> Changing my module to respond to the new "MODULE_DOM_CREATED" notification fixes the problem, but this could have a large impact on any existing modules out there which don't use templating and rely on the DOM_OBJECTS_CREATED notification to start doing whatever they're going to do with the assumption their DOM objects are in place.

shbatm, 2018-01-16:
Thank you for looking into this.

MMM-ip appears to be affected as well.

shbatm, 2018-01-16:
Not sure if this should be a separate issue or not, but it's related to the same commit that originated this bug:
* `this.sendNotification` now fails when called from another module without a `sendTo` parameter:

```js
main.js:88 Uncaught TypeError: Cannot read property 'notificationReceived' of undefined
    at sendNotification (main.js:88)
    at Object.sendNotification (main.js:488)
    at Class.sendNotification (module.js:367)
    at Class.broadcastEvents (calendar.js:529)
    at Class.socketNotificationReceived (calendar.js:106)
    at module.js:245
    at r.<anonymous> (socketclient.js:32)
    at r.emit (index.js:133)
    at r.onevent (socket.js:270)
    at r.MMSocket.self.socket.onevent (socketclient.js:26)
```

MichMich, 2018-01-17:
I merged a pull request yesterday. Did this solve this issue?

ConnorChristie, 2018-01-17:
If it was the PR by me then it wouldn’t fix this second issue, that seems to be a different problem.

@shbatm Would you be able to look into why it’s throwing that exception with the sendTo added?

shbatm, 2018-01-17:
It doesn't appear to be the `sendTo` parameter causing the issue after further investigation. Something with the change to `for (var module of modules) {` from 
```js
for (var m in modules) {
          var module = modules[m]; 
```
Is causing an extra iteration and an `undefined` module variable which causes the error. Reverting to the old method eliminates the problem.

ConnorChristie, 2018-01-17:
Hmm that is interesting. Would you be willing to submit a PR for that change?

shbatm, 2018-01-17:
Will do when I get back to my computer.

shbatm, 2018-01-17:
PR created. I did some more investigating, the problem results from having disabled modules in the config file.

```js
for (var module of modules) {
```
This ES6 method does not behave well with the `modules` array when `config.js` contains disabled modules. It results in an empty `module` variable for every disabled module, which causes an error when the function tries to send a notification to an undefined module.

Reverting back to the original:
```js
for (var m in modules) {
    var module = modules[m];
```

MichMich, 2018-01-19:
Is this issue solved?

shbatm, 2018-01-19:
I believe so. You can close it.

closed 2018-01-19 by MichMich