MagicMirrorOrg/MagicMirror #1142: JQuery UI Touch Punch

faxe1008, 2018-01-15:
**Platform**: Raspbian Jessie Lite, Chromium Webbrowser latest version

**Node Version**: v9.2.1

**MagicMirror Version**:  JavaScript version (v2).

**Description**: I implemented a custom module using http://touchpunch.furf.com/ to make a element which is draggable respond to my interaction with the touchscreen. Problem is that when running the mirror via "npm start" the dragging does not work on the touchscreen. When I open "localhost:8080" in the Chromium Webbrowser it works flawless. 

**Steps to Reproduce**: 
1. Create a blank node module
2. 
`Module.register("MMM-SomeModule", {

    getScripts: function() {
        return [
            "jquery-3.2.1.min.js",
            "jquery-ui.min.js",
            "jquery.ui.touch-punch.min.js"
        ];
    },
    // Override dom generator.
    getDom: function() {

        var wrapper = document.createElement("div");
        var img = document.createElement("img");
        img.src = "SOMEIMAGE.jpg"
       $(img).draggable();
       wrapper.appendChild(img);
        return wrapper;
    }
});`

**Expected Results**: I want to have the dragging working in the actual mirror main window.

**Actual Results**: It does not move when trying to drag.

**Configuration**: Just a normal module import, nothing added.

faxe1008, 2018-01-15:
I created the most basic working example for you to just clone and have a look at it:
[https://github.com/faxe1008/MMM-Issue](https://github.com/faxe1008/MMM-Issue)

MichMich, 2018-01-17:
The issues on GitHub are focussing on the development of the core system. 

Please refer to http://forum.magicmirror.builders for support and troubleshooting when building modules.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-01-17 by MichMich