MagicMirrorOrg/MagicMirror #1144: MMM-Resrobot issue to install , get it work

ayeball, 2018-01-20:
Hej , jag har problem att få mmm resrobot installerat . Har försökt cd/magicmirror https://github.com/Alvinger/MMM-ResRobot.git npm install och även tagit ner filerna och lagt det i modules mappen samt skrivit in i cfg filen med mitt api key samt stationernas id. Inget syns eller funkar . Även skrivit cd/magicmirror/modules https://github.com/Alvinger/MMM-ResRobot.git npm install och den hämtar inte och installerar. Vad glömmer jag / gör för fel ?

Mvh Fredrik

Hi, I'm having trouble getting mmm resrobot installed. Have tried cd / magicmirror https://github.com/Alvinger/MMM-ResRobot.git npm install and also downloaded the files and placed it in the modules folder as well as typed in the cfg file with my api key as well as the stations id. Nothing seems or works. Also wrote cd / magicmirror / modules https://github.com/Alvinger/MMM-ResRobot.git npm install and it does not download and install. What do I forget / make mistakes?

Mvh Fredrik

ayeball, 2018-01-20:
I see it on the screen now but i gott error " MMM-ResRobot : Error: got 400 response
Fetching new departure data for module: MMM-ResRobot
MMM-ResRobot : Error: got 400 response"

MichMich, 2018-01-20:
Hi @ayeball this is an issue with the MMM-ResRobot module. I suggest you open an issue in that repository. Thanks!

Please refer to http://forum.magicmirror.builders for any additional support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-01-20 by MichMich