MagicMirrorOrg/MagicMirror #1145: Help with magic mirror download on raspberry pi3

momoe75, 2018-01-20:
I am new to raspberry and loved everything I read about it.  Thought I give it a try
I just bought a raspberry pi3 which had noobs preloaded.  It allowed me to easily load raspbian.
I updated and upgrade the system based on the instruction that most users used to upload magic mirror.  When it came time to load magic mirror, I input the in the link most used
bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh
Ps: thank you for the short cut
Magic mirror start to load but after a couple seconds, it stops.  There is not error or any sign that something is wrong.  I looks like it has been completed loaded but I know it did not since I can't start it and there is no new MM folder that is made. 

Please help......

Thank youuuuu

MichMich, 2018-01-21:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-01-21 by MichMich

momoe75, 2018-01-22:
Thank you for the response

I followed direction as much as I can and when I try to open it, it says no
such file or directory.  What am I missin?

Thank you

momoe75, 2018-01-22:
Just figured it out

Thank you for your help