MagicMirrorOrg/MagicMirror #1164: error in updatenotification node_helper at line 68

Quarx64, 2018-01-30:
I get an error when MM ist loading third party modules after update to MM 2.2.2

TypeError: Cannot read property 'hash' of null
    at /home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:68:31
    at Git.<anonymous> (/home/pi/MagicMirror/node_modules/simple-git/src/git.js:1482:10)
    at done (/home/pi/MagicMirror/node_modules/simple-git/src/git.js:1354:21)
    at ChildProcess.<anonymous> (/home/pi/MagicMirror/node_modules/simple-git/src/git.js:1379:16)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

i have inserted this
->						if (!err) {
							data.hash = data2.latest.hash;
							self.sendSocketNotification("STATUS", data);
						}
and the error is fixed

Quarx64, 2018-01-31:
oh..., no...
correct is
if (data2.latest) {
  data.hash = data2.latest.hash;
  self.sendSocketNotification("STATUS", data);
}
cause there is no err, data2.latest was null

MichMich, 2018-02-01:
Are you sure this is a core issue? Might be a problem with one of the modules. Dit you install all your modules using git?

Quarx64, 2018-02-01:
I've installed all modules using git. Maybe the attribute -latest is not available on git for the 3rd-party module MMM-Fuel.

MichMich, 2018-02-01:
Could you try removing the `MMM-Fuel` and see if it solves the issue. If you, could you please create an issue in the `MMM-Fuel` repository?

Quarx64, 2018-02-01:
all modules disabled, only defaults are running. the same error.

MichMich, 2018-02-01:
How did you disable the modules? In this case you need to remove the `MMM-Fuel` folder from you `modules` folder as well.

fewieden, 2018-02-01:
If it is an issue with my module, create an issue as @MichMich said, also check if it is just this module or any 3rd party module

Quarx64, 2018-02-01:
@fewieden: MMM-Fuel funktioniert.
After inserting
if (!data2.latest) {
  console.log("Error at module: " + data.module);
}
in updatenotification ->node_helper.js, i got  Error at module: default
hmmm, where is the mistake?

MichMich, 2018-02-01:
That's just a bandage for an issue in `MMM-Fuel`. Please open an issue in the `MMM-Fuel` repo. If @fewieden discovers this really is an issue in the Core, he can send a PR to this Repo.

Closing this issue for now.

Quarx64, 2018-02-01:
02-01-2018 15:07:46 Value for data2.latest in module default found (20823bfc87f387b3d29a30815da9d72e6300c0a9)

02-01-2018 15:10:39 No value for data2.latest in module MMM-WunderGround found (data2.latest is null)
02-01-2018 15:10:39 Value for data2.latest in module MMM-DHT-Sensor found (67036c4acb1d0fa933c3ff5b21c84fadebc2fee9)
02-01-2018 15:10:40 Value for data2.latest in module MMM-SoccerLiveScore found (6ddb0fcb9a8b7ac41ea6922a875486bced9b6273)
02-01-2018 15:10:40 Value for data2.latest in module MMM-Sounds found (94a0bc5b7ff2bd620dccd027481fa64f4b1f3fd7)
02-01-2018 15:10:40 Value for data2.latest in module MMM-Mail found (c24f973314fe8f34af3cfce61001b35727ed021f)
02-01-2018 15:10:40 No value for data2.latest in module default found (data2.latest is null)
02-01-2018 15:10:40 Value for data2.latest in module default found (20823bfc87f387b3d29a30815da9d72e6300c0a9)
02-01-2018 15:10:42 Value for data2.latest in module MMM-Fuel found (4b42070bc0f38df7bf8d87c29b4711c38bab68bb)

02-01-2018 15:20:41 No value for data2.latest in module default found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module MMM-Sounds found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module MMM-WunderGround found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module MMM-DHT-Sensor found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module MMM-SoccerLiveScore found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module MMM-Mail found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module MMM-DHT-Sensor found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module default found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module MMM-Sounds found (data2.latest is null)
02-01-2018 15:20:41 No value for data2.latest in module MMM-WunderGround found (data2.latest is null)
02-01-2018 15:20:41 Value for data2.latest in module MMM-SoccerLiveScore found (6ddb0fcb9a8b7ac41ea6922a875486bced9b6273)
02-01-2018 15:20:41 Value for data2.latest in module MMM-Mail found (c24f973314fe8f34af3cfce61001b35727ed021f)
02-01-2018 15:20:42 Value for data2.latest in module MMM-Fuel found (4b42070bc0f38df7bf8d87c29b4711c38bab68bb)
02-01-2018 15:20:42 Value for data2.latest in module MMM-Fuel found (4b42070bc0f38df7bf8d87c29b4711c38bab68bb)

02-01-2018 15:30:41 No value for data2.latest in module MMM-DHT-Sensor found (data2.latest is null)
02-01-2018 15:30:41 Value for data2.latest in module MMM-Sounds found (94a0bc5b7ff2bd620dccd027481fa64f4b1f3fd7)
02-01-2018 15:30:41 No value for data2.latest in module MMM-WunderGround found (data2.latest is null)
02-01-2018 15:30:41 No value for data2.latest in module default found (data2.latest is null)
02-01-2018 15:30:41 No value for data2.latest in module MMM-SoccerLiveScore found (data2.latest is null)
02-01-2018 15:30:41 No value for data2.latest in module MMM-Mail found (data2.latest is null)
02-01-2018 15:30:41 No value for data2.latest in module MMM-WunderGround found (data2.latest is null)
02-01-2018 15:30:41 No value for data2.latest in module MMM-Sounds found (data2.latest is null)
02-01-2018 15:30:41 No value for data2.latest in module MMM-DHT-Sensor found (data2.latest is null)
02-01-2018 15:30:41 No value for data2.latest in module MMM-SoccerLiveScore found (data2.latest is null)
02-01-2018 15:30:41 No value for data2.latest in module default found (data2.latest is null)
02-01-2018 15:30:41 Value for data2.latest in module MMM-Mail found (c24f973314fe8f34af3cfce61001b35727ed021f)
02-01-2018 15:30:42 Value for data2.latest in module MMM-Fuel found (4b42070bc0f38df7bf8d87c29b4711c38bab68bb)
02-01-2018 15:30:42 No value for data2.latest in module MMM-Fuel found (data2.latest is null)

02-01-2018 15:40:41 Value for data2.latest in module MMM-Sounds found (94a0bc5b7ff2bd620dccd027481fa64f4b1f3fd7)
02-01-2018 15:40:41 Value for data2.latest in module MMM-DHT-Sensor found (67036c4acb1d0fa933c3ff5b21c84fadebc2fee9)
02-01-2018 15:40:42 Value for data2.latest in module MMM-WunderGround found (872cacee5b2adbd84d8478739b1a205f2d8f47b8)
02-01-2018 15:40:42 No value for data2.latest in module MMM-WunderGround found (data2.latest is null)
02-01-2018 15:40:42 No value for data2.latest in module MMM-Mail found (data2.latest is null)
02-01-2018 15:40:42 No value for data2.latest in module MMM-Sounds found (data2.latest is null)
02-01-2018 15:40:42 No value for data2.latest in module MMM-SoccerLiveScore found (data2.latest is null)
02-01-2018 15:40:42 No value for data2.latest in module MMM-SoccerLiveScore found (data2.latest is null)
02-01-2018 15:40:42 No value for data2.latest in module MMM-DHT-Sensor found (data2.latest is null)
02-01-2018 15:40:42 No value for data2.latest in module MMM-Mail found (data2.latest is null)
02-01-2018 15:40:42 No value for data2.latest in module default found (data2.latest is null)
02-01-2018 15:40:42 Value for data2.latest in module default found (20823bfc87f387b3d29a30815da9d72e6300c0a9)
02-01-2018 15:40:43 Value for data2.latest in module MMM-Fuel found (4b42070bc0f38df7bf8d87c29b4711c38bab68bb)
02-01-2018 15:40:43 Value for data2.latest in module MMM-Fuel found (4b42070bc0f38df7bf8d87c29b4711c38bab68bb)

MichMich, 2018-02-01:
Is this in your terminal console or in the web-inspector?

Quarx64, 2018-02-01:
It's from the terminal console and logged in the pm2 mm-out-0.log.
Could be a git error. No idea. To many requests in a short time?

MichMich, 2018-02-01:
I'm running the new version, and I'm not seeing any issues. I didn't get any other other mentions of this issue since the release of the latest version.

I suggest you backup your config file, custom css file and modules folder and reinstall MM to see if that solves the issue.

MichMich, 2018-02-03:
@Quarx64 Any results?

Quarx64, 2018-02-08:
After new install, no more errors.

MichMich, 2018-02-08:
Great! Thanks for the feedback!

closed 2018-02-08 by MichMich