MagicMirrorOrg/MagicMirror #1168: Config getRelative and urgency seem to do the same thing (wrongly)

E3V3A, 2018-02-02:
According to the README description the config options" `getRelative` and `urgency` are doing same thing. While both those things are not very productive friendly. 

**What should happen?**

*  Many days (let's say >7) before the event, one would like to see relative times in the form `In 8 days`.  
* A few days (let's say <7) before the event, one would like to see relative times in the form `In 8 days @ 15:00`.
* Two days before the event, one would like to see relative times in the form `Tomorrow @ 15:00`.
* At the *same day*, one would like to see **absolute** times in the form:
  - `Today @ 15:00`.
  - `In 15 minutes @ 15:00`

**What is actually happening? (default settings)**

* My time is showing `09:35` and the event only reads `In 2 hours`. Totally useless, since, I don't have any event at `11:35`!

E3V3A, 2018-02-03:
@MichMich To make a PR we first need to understand what was intended with those options. 

To fix the other details, we can figure that out later/after.

MichMich, 2018-02-03:
Ok, let's take a different approach: are you running into issues? Are you getting incorrect calendar notifications. Is it an issue which you can replicate in a small prove of concept. (Try out the two options to see the difference and see how they overlap.)

If so, please fill in the full issue report including all asked questions. 

The current description doesn't really explain the issue or the expected result.

MichMich, 2018-03-25:
Closed due to inactivity (and not pointing out any real issues ... ;))

closed 2018-03-25 by MichMich