MagicMirrorOrg/MagicMirror #1174: Discussion: Bundle MagicMirror package to distribute more easily

ConnorChristie, 2018-02-07:
Electron provides a nice way to produce executables and packages for modules such as MagicMirror. Some benefits this would provide to users wanting to install MagicMirror are listed below.

Benefits:
- Faster install time (no need to `npm install`)
- More user friendly being a bundled "executable" users can install and run out of the box

Changes required:
- [ ] Needs to reference external config file
- [ ] 3rd party modules folder to be referenced externally
- [ ] Add helper commands to install modules more easily?

I know most of the current user base is more advanced and they know how to use git and whatnot but going forward into the broader community where users don't know how to pull down the latest changes or fetch modules into the module folder would be more friendly to them. I feel this would increase the adoption rate significantly and lower the barrier to entry for the MagicMirror space which would allow a vast number of users to enter it more easily.

Let me know your thoughts and the pros and cons that come with it.

MichMich, 2018-02-07:
@ConnorChristie I appreciate your suggestion.

The only issue I have with it, is that it might make the project a bit overcomplicated. The reason I say this, is because I started this project as an ultimate starter project for Raspberry Pi enthousiasts. As a matter of fact, for most of the contributors, the MagicMirror project is the first open source project they ever contributed to. This is one of the reasons why the MagicMirror project is featured in several RasPi magazines.

The project has a lot of opportunities for improvement. We could use a powerful framework like Vue to ramp up the development speed. We could use SASS for better/easier css implementations. We could make it an NPM installable package. And as you say, we could bundle it up. The big downside of of of these changes is that it overcomplicates things: a user no longer will be able to open just one file and make a small modification and see how it works out.

Of course, a bundled version can be complimentary to the regular un-bundled version. And I'm sure a lot of (new) users will opt for the bundled version. But this means those users won't be motivated to take a peek under the hood. They will just remain 'users'. They won't become contributors, and worse: they won't be motivated to take their first steps in software development.

And to be honest: motivating curious users to step out of their comfort zone and take those first steps is what drives me in this project. Therefor my ultimate goal is this project is to keep it as accessible as possible.

Note: I'm aware this explanation should probably be part of some manifesto within the project. I'll put it on my TODO list. All the way at the bottom. ;)

MichMich, 2018-02-07:
I took the easy route out: https://github.com/MichMich/MagicMirror/tree/develop#manifesto ;)

ConnorChristie, 2018-02-07:
One thing that was the main driver behind this idea is the long time it takes to install MagicMirror on the different platforms. With packaging it, the user wouldn't have to use npms slow way of downloading dependencies.

I also think the benefits would still be there if we were to package it and make it an option for users that may want to set it and forget it. It is truly nice giving everyone the ability to be able to easily contribute to this open source project but one use case I can think of is that this might inhibit people who would be deploying to a mass number of MagicMirrors when you don't need the source code deployed to each RPi but rather want a stable release.

rejas, 2018-02-17:
I have no clue on what would have to doon to achiev running mm in an electron app (not familiar with the mm sources nor the electron app). 
but looking from a project point of view, asking the both of you from their perspective, would it be possible to have a "sister"-project that pulls in the mm sources and builds an electron app out of it somehow (semi)automatically?

E3V3A, 2018-03-05:
@ConnorChristie  

> ... but one use case I can think of is that this might inhibit people who would be deploying to a mass number of MagicMirrors when you don't need the source code deployed to each RPi but rather want a stable release.

That is called commercialization. And from a mass perspective and my own experience, such projects rarely contribute to the FOSS development community, except for getting their own problems solved. They just want to make quick money on other peoples efforts. In fact what you describe, has already been done by a company, who seem to have repackaged this project and turned into a business. 

So why should "we" bother making it easier for them, with little in return? After all, anyone can still fork and bundle this as they wish. Knock yourself out! :)

rejas, 2018-03-25:
@E3V3A who is that company and did anyone try to contact them / talk to them about giving back to the community?

MichMich, 2018-03-25:
I'n not seeing this discussion going anyway. I made my point why I personally am not going to integrate it. Not saying I won't accept any PR's. But for now I close this issue. This discussion is better suited on the forum.

closed 2018-03-25 by MichMich