MagicMirrorOrg/MagicMirror #1181: Documentation: diagrammatic information flow

E3V3A, 2018-02-14:
The learning curve for implementing new modules for non-developers is quite long, if not steep. Some additional but not too technical documentation for beginners would probably be very helpful. 

For example, if someone who have a good understanding of the MM structure, might consider making ~2-3 diagrams and flow charts about how the MM framework is related to its various parts, like the modules, the server, the browser, the npm imports etc. 

For example, a picture **conceptually similar** to (this totally unrelated) diagram: 

---

[image]

---

This would hopefully help people to better understand how to go from an idea of having a blob of data, to getting it displayed in an MM Module. 

I think it might also help people troubleshoot their modules, by allowing them to more clearly speak about (and isolate) what parts are causing them trouble. (I.e. Is it with the data collection, the data presentation or the socket communication, etc?)

MichMich, 2018-02-14:
Hi @E3V3A, I absolutely appreciate your contributions, but so far I don't think raising issues like these (and the few others you recently created like the moment.js framework) are having a positive impact on the clarity of the repositories issues list. I generates some cluttering. Suggestions like these are absolutely valuable, but better suited on the forum.

So for future issue, please keep the following in mind:
- If there is an issue you can solve yourself, a PR if preferred over an Issue.
- Suggestions, thoughts, improvement ideas (like this issue) should go on the forum or on slack.
- Replicable bugs can be reported using the GitHub repository issue lists, but should in all cases contain every requested detail regarding versions, results and desired results. 

This is the only way I can keep a good track of the real problems and issues of the framework.

Thanks for your understanding! Keep up the great work and contributions.

closed 2018-02-14 by MichMich

E3V3A, 2018-02-14:
@MichMich  No problem. Roger that.