MagicMirrorOrg/MagicMirror #1187: Auto install fails to install dependencies

joncage, 2018-02-26:
**Platform**: Raspberry Pi 3

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Latest from github as of 26/02/18

**Description**: Fresh install of noobs on a brand new Raspberry Pi 3. Ran the auto install script specified from `readme.md ` and it failed.

**Steps to Reproduce**: Run the auto install command:

```
bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"
```

...then wait for the install to fail with the following message:

```
Cloning MagicMirror Done!
Installing dependencies ...
npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.
npm WARN prefer global coffeescript@1.10.0 should be installed with -g
npm WARN prefer global jsonlint@1.6.2 should be installed with -g

> uws@9.14.0 install /home/pi/MagicMirror/node_modules/uws
> node-gyp rebuild > build_log.txt 2>&1 || exit 0
```

**Expected Results**: A message indicating the install completed as expected.

**Actual Results**: The error message above.

**Configuration**: No idea; Didn't get that far.

MichMich, 2018-02-27:
Did you just get those warnings? Or does it fail to start? Because those warnings kan be ignored.

joncage, 2018-02-27:
I did get it to start but the page it displays is completely black with no text so I had assumed something didn't work?

ibrahim5alil, 2018-03-05:
please guys every time i try to use the auto installation code its Dosn't take any action how i can solve this 

pi@raspberrypi:~ $ curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash
pi@raspberrypi:~ $

fewieden, 2018-03-05:
@ibrahim5alil using a non outdated cmd would be a good start ^^

```bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"```

ibrahim5alil, 2018-03-05:
@fewieden  i'v tried both of the codes and it didn't work 
can u explain wt do u mean by using a non outdated cmd please

MichMich, 2018-03-05:
What happens if you leave out the `| bash` in the first command?

ibrahim5alil, 2018-03-05:
@MichMich nothing just no action at all
pi@raspberrypi:~ $bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"
pi@raspberrypi:~ $

MichMich, 2018-03-05:
No, the first one:
`curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh`

ibrahim5alil, 2018-03-05:
pi@raspberrypi:~ $ curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash
pi@raspberrypi:~ $ curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash
pi@raspberrypi:~ $ curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh
pi@raspberrypi:~ $ bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"
pi@raspberrypi:~ $ curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh
pi@raspberrypi:~ $ 

@MichMich  just no action

MichMich, 2018-03-05:
are you sure your pi is connected to the internet?

ibrahim5alil, 2018-03-05:
@MichMich  yes i can clone and update i can enter any code except this
[image]

MichMich, 2018-03-05:
What happens if you run `curl google.com`?

ibrahim5alil, 2018-03-05:
[image]
@MichMich

MichMich, 2018-03-05:
Try `curl https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh`

ibrahim5alil, 2018-03-05:
[image]
@MichMich nothing

MichMich, 2018-03-05:
It appears the your pi cant resolve the GitHub domain name. Which DNS are you using?

ibrahim5alil, 2018-03-05:
[image]
@MichMich

MichMich, 2018-03-05:
That’s your router. You might want to try a direct DNS like googles. 8.8.8.8

Furthermore this isn’t a MagicMirror issue. It’s more a generic Pi issue. I suggest you try to get some help on the forum http://forum.magicmirror.builders

ibrahim5alil, 2018-03-05:
thank you @MichMich for you helping me

MichMich, 2018-03-05:
You’re welcome!

closed 2018-03-05 by MichMich

braunprinz, 2019-02-26:
Hello guys,
I was trying to install MagicMirror, I got the error message
"Unable to Install dependencies"
How can I solve this
I tried to install it few times, even manually it didn't work
[image]

braunprinz, 2019-02-26:
Got it, the npm package was not installed
Just run the following command
sudo apt-get install npm

braunprinz, 2019-02-26:
I have tried to install npm package. After doing that i run the auto,atic installation again, and I got the same error message
[image]