MagicMirrorOrg/MagicMirror #1189: Google calendar not loading.

tadeubanzato, 2018-03-01:
It does not show any error, the curl from the calendar URL is working but the content is not loading into the magic mirror.
Been trying everything without success.

Seems that this started to happen when Google changed it's interface and even setting the calendar to public it doesn't load on the mirror.

E3V3A, 2018-03-01:
Hi! @tadeubanzato Please ask the question with much more details and use the forums. 
Details for using various calendar functionality can be found [in the Wiki here](https://github.com/MichMich/MagicMirror/wiki/Apple-Google-MicroSoft-Calendar-Support).

tadeubanzato, 2018-03-01:
Thanks @E3V3A  but that didn't help.
I'm not looking to sync more than one calendar, the issue is that the events from the public calendar from Google itself is not loading on the Magic Mirror.
The public calendar test I'm trying to load on the Mirror is this one:
https://calendar.google.com/calendar/embed?src=83j7tsjdivl1q2mquioftrnqsk%40group.calendar.google.com&ctz=America%2FSao_Paulo

It keeps showing me "No Upcoming Events" even with the events on the callendar

When I run curl https://calendar.google.com/calendar/embed?src=83j7tsjdivl1q2mquioftrnqsk%40group.calendar.google.com&ctz=America%2FSao_Paulo

E3V3A, 2018-03-02:
Maybe the events are marked as private? Anyway, I've been told to keep these discussion to the forum, so let's do that.

MichMich, 2018-03-06:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-03-06 by MichMich

Amyiam, 2019-01-09:
I'm also getting no upcoming events. 
Will appreciate the help thanks