MagicMirrorOrg/MagicMirror #1192: Turning on this plug in breaks showing the height/width of objects

Camicdev, 2018-03-02:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Google Chrome, MacBook Pro (Newest), The mac version before Sierra

**Node Version**: Uhhh Newest sketch version?

**MagicMirror Version**: 3.52

**Description**: If I turn on this plug in, when I click on objects, the height and width fields are empty

**Steps to Reproduce**: ^

**Expected Results**: Sizes

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

MichMich, 2018-03-02:
Which module?

MichMich, 2018-03-06:
Please open een issue in the module's repository.

closed 2018-03-06 by MichMich