MagicMirrorOrg/MagicMirror #1196: Calendar and news feed stuck loading under language:"zh_cn"

philipdeng, 2018-03-06:
MM works well under all English environment. But it does not work if calendar or newsfeed contain any Chinese characters
However, after changing to language:"zh_cn" in config file, calendar and newsfeed stuck loading even it could be displayed under language en setting.
[image]
[image]

MichMich, 2018-03-25:
Did you check the encoding property you can set for news feeds?

philipdeng, 2018-04-03:
Yes, I tried all available encoding, but not work.

philipdeng, 2018-04-04:
It will be stuck on loading if this calendar or newsfeed contain any Chinese characters。

fewieden, 2018-04-04:
Could you provide the config for your newsfeed?

WuWeiWW, 2018-06-26:
这个是因为 用了`moment` 组件，设置本地语言时候 `setLccation`或 `updateLocal` 时候需要 `zh-cn` 但是传入的是 `zh_cn`;
解决方案: 修改 `translations.js` 和 `config.js` `zh_cn`为 `zh-cn`

MichMich, 2018-06-27:
Merged.

closed 2018-06-27 by MichMich