MagicMirrorOrg/MagicMirror #1204: Failed to autostart on Raspberry Pi 3

ericchenkai, 2018-03-12:
I installed Nodejs 9.8.0 on my raspberry Pi 3 model B before run the installation script for raspberry.  
MagicMirror does not autostart until I found a fix on internet: 
`sudo env PATH=$PATH:/usr/local/bin pm2 startup systemd -u pi --hp /home/pi`

E3V3A, 2018-03-13:
Did you install pm2 with sudo? (where?)
Are you sure you wanna run MM as root?

ericchenkai, 2018-03-15:
@E3V3A I installed MagicMirror2 the easy way with below script on my new raspberry Pi 3. 
https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh
```
read -p "Do you want use pm2 for auto starting of your MagicMirror (y/n)?" choice
if [[ $choice =~ ^[Yy]$ ]]; then
    sudo npm install -g pm2
    sudo su -c "env PATH=$PATH:/usr/bin pm2 startup linux -u pi --hp /home/pi"
    pm2 start ~/MagicMirror/installers/pm2_MagicMirror.json
    pm2 save
fi
```

Regarding run MM as root, no, I won't. 
I dropped my first installation of MM and manually installed MM again and created a systemd service to auto start MM `serveronly` mode. 
```
cat << EOL | sudo tee /etc/systemd/system/magicmirror.service
[Unit]
Description=Magic Mirror
After=network-online.service

[Service]
Environment=NODE_PORT=8234
Type=simple
User=$USER
WorkingDirectory=$HOME/MagicMirror/
ExecStart=/usr/bin/node serveronly/

[Install]
WantedBy=multi-user.target

EOL

sudo systemctl --system daemon-reload
sudo systemctl enable magicmirror
```

ericchenkai, 2018-03-15:
This issue has solved to me. So it is time to close this topic.

closed 2018-03-15 by ericchenkai