MagicMirrorOrg/MagicMirror #1210: What is the current status for dependecies?

E3V3A, 2018-03-14:
What is the current dependency and compatibility status for MM?
Can I safely update these?

```bash
# nvers
v9.8.0
5.7.1

# npm-check-updates
[INFO]: You can also use ncu as an alias
Using /home/pi/MagicMirror/package.json
[..................] - :
 electron   1.4.15  →   1.8.3
 mocha      ^4.1.0  →  ^5.0.4
 spectron    3.7.x  →   3.8.x
 stylelint  ^8.4.0  →  ^9.1.2

The following dependencies are satisfied by their declared version range, but the installed versions are behind. You can install the latest versions without modifying your package file by using npm update. If you want to update the dependencies in your package file anyway, run ncu -a.

 colors               ^1.1.2  →   ^1.2.1
 express             ^4.16.2  →  ^4.16.3
 helmet               ^3.9.0  →  ^3.12.0
 request             ^2.83.0  →  ^2.85.0
 rrule-alt            ^2.2.7  →   ^2.2.8
 simple-git          ^1.85.0  →  ^1.92.0
 grunt-markdownlint  ^1.0.43  →   ^1.1.0

Run ncu with -u to upgrade package.json
```

I read somewhere that *electron* should be kept back, is this still true?

E3V3A, 2018-03-15:
This is also related to closed issues #145 and #150. If all ok, we need to update README.

MichMich, 2018-03-25:
We could update them, but we need to do some testing. Since I'll be releasing the next version within a few days, it's better to postpone this.

justjim1220, 2018-03-25:
Will the next version be able to update the current version we already have installed without having to replace all our changes with modules added and such?

MichMich, 2018-03-25:
Yes. One simple command without breaking changes. I release a new version every three months.

Kiina, 2018-03-25:
Will the update to electron 1.7 be included? The downgrade broke my module but on the dev channel the changelog lists it as change in 2.2.2 which was released without this change on the master branch.

E3V3A, 2018-03-26:
I think the biggest (and most risky) change is with *electron* from: `1.4.15  →   1.8.3`. However, because of the very wide use of electron, I'd also be surprised if it broke something serious. I'd guess we'd have heard of it already. But then, perhaps other deps are requiring lower versions?

So if someone running MM on an RPi, and have successfully updated this already, then please let us know!

E3V3A, 2018-03-27:
I think this should be a **high priority** issue, now that I see that *electron* has a [Remote Code Execution vulnerability](http://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2018-1000006) in it. What this means, is that we risk to turn every single MM into a bot-net, or worse, a root leverage point in the local network of anyone using it. 

Basically you can gain a command shell from [something like](https://www.exploit-db.com/exploits/43899/) this:

```js
<!doctype html>
<script>
  window.location = 'exodus://aaaaaaaaa" --gpu-launcher="cmd" --aaaaa='
</script>
```

In view of this, we should consider adding one or both of the following dependency tags, in the top README.

https://gemnasium.com/github.com/MichMich/MagicMirror
https://david-dm.org/MichMich/MagicMirror

MichMich, 2018-03-27:
I’m aware of that. But keep in mind that electron (in this case) isn’t use as a browser. The only way external code could enter Electron is by a third party module. But modules can already execute commands using the node helper. 

Btw, in the dev branch Electron is already (more) up to date.

E3V3A, 2018-03-27:
Fantastic. Good to hear. But perhaps still good to know for other users?

Kiina, 2018-03-27:
Well the one on dev branch is affected too but I don't really see much reason why shipping the next version with 1.7.13 instead of 1.7.10 would cause any problems. Some modules use it as a browser so theoretically it could happen, but its a rather small attack vector

MichMich, 2018-03-27:
Feel free to send a PR with an updated version. 👍🏻

MichMich, 2018-03-28:
Merged: https://github.com/MichMich/MagicMirror/pull/1232

MichMich, 2018-04-01:
So ... that was "fun": https://github.com/MichMich/MagicMirror/issues/1243

MichMich, 2018-04-02:
I think we need to investigate how we get the new version of Electron to work on a Pi.

Kiina, 2018-04-02:
Was noone running the dev branch on a raspberry? Because there I never saw
any complaints about it not working... Really weird

MichMich, 2018-04-06:
De dev branch is now updated to Electron 2 beta. Hopefully Electron 2 is out of beta before the next release.

closed 2018-04-06 by MichMich