MagicMirrorOrg/MagicMirror #1211: mm not opening

specialsnow27, 2018-03-15:
just got a fresh install of raspbian on my RPI3B and after i installed magicmirror and typed in the terminal `npm start` it showed me this long line of errors 

`> magicmirror@2.2.2 start /home/pi/MagicMirror
> sh run-start.sh

/home/pi/MagicMirror/node_modules/electron/index.js:9
  throw new Error('Electron failed to install correctly, please delete node_modules/' + path.basename(__dirname) + ' and try installing again')
  ^

Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/index.js:9:9)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.require (module.js:504:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/cli.js:3:16)
    at Module._compile (module.js:577:32)

npm ERR! Linux 4.4.50-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v6.13.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.2.2 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.2.2 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
`

if anyone can help me get rid off this problem i will be grateful
thx

MichMich, 2018-03-15:
Try `npm install` in the MagicMirror folder.

specialsnow27, 2018-03-15:
@MichMich just did it and after it finished wrote `npm star` and it showed the same errors

MichMich, 2018-03-15:
It seems there is an issue with Electron so I suggest you open an issue in the electron repository. It might be due to the fact you are using the new `RPI3B` which might not yet be supported by Electron.

MichMich, 2018-03-15:
Just checking: are you using the 3B or the 3B+?

specialsnow27, 2018-03-15:
@MichMich thx for the help. im using a 3B 
and sorry for asking but what is electron?

MichMich, 2018-03-15:
Electron is the browser that is used to show the application.

What OS are you running? Jessie? Jessie Lite?

specialsnow27, 2018-03-15:
@MichMich i have no idea how can i check?

specialsnow27, 2018-03-15:
@MichMich i checked and im pretty sure i have jessie

MichMich, 2018-03-15:
You might have jessie light.

I recommend doing a fresh install for Jessie (Full Edition) and then run the MM installer.

specialsnow27, 2018-03-15:
@MichMich i just finished checking and i have full jessie

specialsnow27, 2018-03-15:
still doesn't work

E3V3A, 2018-03-16:
@specialsnow27 Do this:

```bash
uname -a
sudo lsb_release -a
dmesg |grep OF:
```

The paste the results in within triple back-ticks (```) here.

closed 2018-03-16 by specialsnow27