MagicMirrorOrg/MagicMirror #1222: Configuration Of NewsFeed Module

Linsanity77, 2018-03-22:
Hello,
I am trying to customize the modules shown on my MagicMirror. I am trying this through editing  config/config.js.
In the newsfeed module, how can I make it so that it won't show news from a long time ago? I think there is ignoreOlderThan and ignoreOldItems but how would I add those into the module?
This is what the module looks like:
 {
                        module: "newsfeed",
                        position: "bottom_bar",
                        config: {
                                feeds: [

                                        {
                                                title: "New York Times",
                                                url: "http://www.nytimes.com/se$
                                        },
                                        {
                                               title: "CNN",
                                               url: 'http://rss.cnn.com/rss/cnn_topstories.rss'
                                        },
                                ],
                                showSourceTitle: true,
                                showPublishDate: true
                        }
                },
        ]

};
Thanks.

justjim1220, 2018-03-22:
Look in the newsfeed.js file. Edit it there...

Or, there are config perimeters in there to use in the config.js file.

justjim1220, 2018-03-22:
// Default module config.
	defaults: {
		feeds: [
			{
				title: "New York Times",
				url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml",
				encoding: "UTF-8" //ISO-8859-1
			}
		],
		showSourceTitle: true,
		showPublishDate: true,
		showDescription: false,
		wrapTitle: true,
		wrapDescription: true,
		truncDescription: true,
		lengthDescription: 400,
		hideLoading: false,
		reloadInterval: 5 * 60 * 1000, // every 5 minutes
		updateInterval: 10 * 1000,
		animationSpeed: 2.5 * 1000,
		maxNewsItems: 0, // 0 for unlimited
		ignoreOldItems: false,
		ignoreOlderThan: 24 * 60 * 60 * 1000, // 1 day
		removeStartTags: "",
		removeEndTags: "",
		startTags: [],
		endTags: [],
		prohibitedWords: []
	},

Linsanity77, 2018-03-22:
Thanks, but uh how do I get to newsfeed.js?
Sorry, I'm new to this.

justjim1220, 2018-03-22:
MagicMirror/modules/default/newsfeed/newsfeed.js

Linsanity77, 2018-03-22:
Do I run it in terminal?

justjim1220, 2018-03-22:
I'm a newbie as well, still having problems with a lot of the modules I want to try. But, I did figure this out a while back. Actually glad I was able to help someone instead of being the one needing help! LOL!!!

Linsanity77, 2018-03-22:
Never mind I found it, but just wondering, can you get to it through terminal or do you have to go to it through files?

Linsanity77, 2018-03-22:
Oh

Linsanity77, 2018-03-22:
Lol, thanks so much though, you helped out a lot thanks!

justjim1220, 2018-03-22:
open it with your regular editor just like you do with the config.js file

Linsanity77, 2018-03-22:
Sorry, how's that? I use sudo nano config/config.js to open it.

Linsanity77, 2018-03-22:
to open the config.js

Linsanity77, 2018-03-22:
Anddd, opening the newsfeed.js gives me the New York times feed but what about ones I added?
Sorry about all the questions, I just want to make sure I'll understand what I'm doing. Thanks

justjim1220, 2018-03-22:
make sure you save any changes you make to any of the css or js files.

I've also found that when I make changes, things run a bit smoother if you rebuild before running the mirror.

sudo nano /modules/default/newsfeed/newsfeed.js

justjim1220, 2018-03-22:
I'm currently stuck on the Calendar Module. Trying to get it to pull from my Google Calendar... UGH!!!

Linsanity77, 2018-03-22:
What do you mean by rebuilding?

Linsanity77, 2018-03-22:
Also, Idk if this will help but something I was using to help me set it up had something on the calendar. 
He wrote:
For now, I decided to remove both and instead subscribe to the Tampa Bay Rays schedule and a Google Calendar containing a list of national holidays. These calendar URLs normally end in .ics, and you can Google the one you're looking for.

Down the road, I'll add authentication so I can link up my Google Calendar. You can also change the icon that's loaded by changing the value of calendar.urls.symbol; this value should be from the list of FontAwesome icons found here, with the "fa-" portion removed. For example, 'fa-bullhorn' becomes just 'bullhorn'.

justjim1220, 2018-03-22:
after you add a module, you should cd to that directory and install it
I use Node.js so my install is npm install.
then you should configure the config.js file to the defaults and perimeters you need.
then I go back to the MagicMirror directory and npm rebuild.
then I run the mirror by using npm start
.

Linsanity77, 2018-03-22:
uhhhhhh

justjim1220, 2018-03-22:
sorry, took so long to answer, had a customer stop by and will need a few to repair his phone. I'll try to answer when I can. And, thanks for the Calendar info, hope it will help me a bit.

Linsanity77, 2018-03-22:
It's fine and no problem

justjim1220, 2018-03-22:
I am probably doing mine differently than 95% of all those doing this, I'm running it on Windows 10...

Linsanity77, 2018-03-22:
Yeah, you completely lost me with the rebuild instructions. lol sry, is it for when I'm adding like a new thing to the interface, like the weather for example if I didn't have it before? Also, do you have any recommendations for a module I should TRY to install?

Linsanity77, 2018-03-22:
Ohhhhhhh. I'll do my best to try to install new modules. Yeah, I'm using the raspberry pi, but it's mostly the same I think.as in the newsfeed.js worked.

Linsanity77, 2018-03-22:
How are you using Windows 10 to do it though? Are you linking your computer to the monitor? I'm pretty curious because I didn't know it could be done with a different system.

justjim1220, 2018-03-22:
I started by getting all the default modules to work, then I worked on customizing them to my preferences. Have tried a few others to get myself aquainted with the process, I have the MMM-SimpleLogo working. I have the Weather working for where I live. I changed the Comments to my own phrases, and I have the newsfeed to show my local headlines

Have been trying to get MMM-alexa and MMM-awesome-alexa, and the google assistant to all work without any luick. As well as the Calendar, wanting it to bring up my personal events.

A few others, but I need to get this phone done, LOL. holler at ya later!

Linsanity77, 2018-03-22:
Wow, you're pretty accomplished. That's cool. For the Alexa part, I previously worked on the AlexaPi project. I was successful, but I don't know if it will still work with the MagicMirror. It's pretty cool though, the AlexaPi if you want to give it a shot or as a seperate project.
It took me a couple tries though lol. If you want to delve into it, here's the link I used that helped me build/program it: https://lifehacker.com/the-simplest-way-to-build-a-raspberry-pi-powered-amazon-1794218212

Linsanity77, 2018-03-22:
*facepalm* I realized you said you used Windows 10. Sorry, the AlexaPi is for the Raspberry PI, BUT maybe you'll want to try it? I got mine as a gift but I heard it's quite inexpensive for a mini computer.

justjim1220, 2018-03-22:
I would love to have a Raspberry Pi!!! Just can't come up with the extra money to get one.
Phone is done, so, got a few minutes

justjim1220, 2018-03-22:
most everything is coded pretty much similar, I just have to use different methods to convert raspi code to work with Windows.
I will definitely look into the site you passed along my way, hopefully it will help me in a great way.
I have a degree in Computer Programming, got it in 2003, but haven't used it since graduating. I also have a degree in Computer Networking and MicroComputer Specialist.
I am a forced retired registered nurse of 20+ years, did the computer schooling for something to do cause it was interesting. This coding isn't anything like I've learned back then, but it is similar enough for me to figure it out.

justjim1220, 2018-03-22:
So, my first advice is to remember to be in the MagicMirror/modules to grab the modules codes. then cd to the module your working on, ie: from the directory> MagicMirror/modules, type cd module name. Make sure it is typed exactly as it is named. ie: cd MMM-alexa. then use your code to install it. for me, it's npm install.

then go back to the MagicMirror directory, for me it's, cd .. enter, cd .. enter, cd .. enter. then rebuild it from the MagicMirror directory. for me its npm rebuild
 then place the code for the module in the config.js file. then run it. for me it's npm start.

once I get it installed, and working, then I start tweaking it to my perimeters.

Linsanity77, 2018-03-22:
Is that for the rebuilding part earlier?

justjim1220, 2018-03-22:
I repair, refurbish, buy, sell, trade, etc. smartphones, tablets, laptops, desktops, and flat panel tv's.

Visual Basic, C++, Cobol, Pascal, Java, HTML, and a slew of others I can't pull out of my brain at the moment!

I had a quintuple coronary bypass in April of 2013, followed by getting Meniere's Disease shortly thereafter. So, I do all of this to keep me busy and from going completely bonkers!!!

justjim1220, 2018-03-22:
Yes

Linsanity77, 2018-03-22:
Cool. You're very versed in computer lingo, that's awesome! Have you ever tried other projects?

justjim1220, 2018-03-22:
Only when I was attending college.
But with all my assignments, I usually went above and beyond the actual
assignment itself. Like, we were just supposed to make simple programs with
certain perimeters, and I would create a full program and package it to
install on computers.

closed 2018-03-25 by MichMich