MagicMirrorOrg/MagicMirror #1228: Calendar Bug? Reccuring fullDayEvent (Birthday) - today is wrong (one day before real date)

eckonator, 2018-03-25:
**Platform**: Raspberry Pi 3, OwnCloud based Calendar
**Node Version**: v6.13.1

**MagicMirror Version**: Latest JavaScript Version (v2).

**Description**: I have a Birthday Calendar stored on my OwnCloud Instance. There are hosted multible Calendards. Dates that are based on a time works well. But full day events (birthdays) that are recurring are shown on my magic mirror one day before as today instead of the real today-date.

**Steps to Reproduce**: Just include this ICS File:

`
BEGIN:VCALENDAR
VERSION:2.0
CALSCALE:GREGORIAN
PRODID:-//SabreDAV//SabreDAV//EN
X-WR-CALNAME:Test-Kalender
X-APPLE-CALENDAR-COLOR:#c274e7
BEGIN:VEVENT
CREATED:20180325T120003
DTSTAMP:20180325T120003
LAST-MODIFIED:20180325T120003
UID:N25F8LCTXDSGTY7VDVEY
SUMMARY:Birthday People 1
CLASS:PUBLIC
STATUS:CONFIRMED
DTSTART;VALUE=DATE:20180326
DTEND;VALUE=DATE:20180327
RRULE:FREQ=YEARLY
END:VEVENT
BEGIN:VEVENT
CREATED:20180325T120033
DTSTAMP:20180325T120033
LAST-MODIFIED:20180325T120033
UID:V7QP0NKOH8MBCAP90JPAK6
SUMMARY:Birthday People 2
CLASS:PUBLIC
STATUS:CONFIRMED
DTSTART;VALUE=DATE:20180327
DTEND;VALUE=DATE:20180328
RRULE:FREQ=YEARLY
END:VEVENT
END:VCALENDAR
`

**Expected Results**: Today output should be on the right day (but it's one day before)

**Configuration**: 

`
{
            module: "calendar",
            header: "Birthdays",
            position: "top_left",
            config: {
                calendars: [
                    {
                        //symbol: "calendar-check-o ",
                        symbol: "birthday-cake ",
                        maximumEntries: '2',
                        maximumNumberOfDays: '14',
                        url: "webcal://link-to-my-calendar.ics"
                    }
                ]
            }
 }
`
Hope someone can help or have a fix for me? This project is so cool :)

eckonator, 2018-03-25:
I think i found the bug, i changed in:

`MagicMirror/modules/default/calendar/calendar.js`

Line 350 from

`event.today = event.startDate >= today && event.startDate < (today + 24 * 60 * 60 * 1000);`
to

`event.today = event.startDate >= (today + 24 * 60 * 60 * 1000) && event.startDate < (today + 24 * 60 * 60 * 1000);`

It fixes my Problem.

MichMich, 2018-03-25:
That doesn't make sense. You are comparing it twice with the same value?

MichMich, 2018-03-25:
Please continue in: https://github.com/MichMich/MagicMirror/issues/1105

closed 2018-03-25 by MichMich