MagicMirrorOrg/MagicMirror #1233: 'fatal' message...

justjim1220, 2018-03-27:
when running my mirror, I get this message popping up in Git Bash...

fatal: Not a git repository (or any of the parent directories): .git

Not sure what to think of it because everything IS running that I have coded in it as it should:

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "", "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	language: "en",
	timeFormat: 24,
	units: "imperial",

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
		        module: 'MMM-SimpleLogo',
		        position: "top_center",    // This can be any of the regions.
		        config: {
		            // The config property is optional.
		            // See 'Configuration options' for more information.
 			       }
 		},
		{
			module: "clock",
			displayType: "both", // options: digital, analog, both
			position: "top_left"
		},
		{
			module: "compliments",
			position: "top_bar"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
					symbol: "calendar-check-o ",
					url: "webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics"
					}
					   ]
				}
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "Bethany",
				locationID: "5054079",  //ID from http://www.openweathermap.org/help/city_list.txt
				appid: "***************************************",
				roundTemp: "true"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "Bethany",
				locationID: "5054079",  //ID from http://www.openweathermap.org/help/city_list.txt
				appid: "***************************************",
				roundTemp: "true"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
					title: "Bethany Local News",
					url: "https://news.google.com/news/rss/local/section/geo/Bethany,%20MO,%20United%20States/Bethany,%20Missouri?ned=us&hl=en&gl=US",
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
		
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

Any ideas would be much appreciated!!!

KilianB, 2018-03-27:
have you accidentally deleted any of the automatically generated files by git? e.g. is there a .git file present in your root project folder `ls -a` ? 
It seems like you have the code in your folder therefore the mirror runs fine but updating will not be possible. Try to clone the repository again. Even though the gitignore should exclude the modules folder just do a backup and start over.

justjim1220, 2018-03-28:
there is a .github directory and a .gitignore file

[image]

I'll try the backup and startover...

thanks

MichMich, 2018-03-28:
It is not a git repository (you probably downloaded is using a zip file). Please instal using git.

Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-03-28 by MichMich