MagicMirrorOrg/MagicMirror #1238: updatenotification module cause excessive spam in pm2 error logs

E3V3A, 2018-03-30:
**Platform**:  RPi3  
**Node Version**:  9.8.0  
**MagicMirror Version**: 2.2.2    

**Description**:   
[updatenotification](https://github.com/MichMich/MagicMirror/tree/develop/modules/default/updatenotification) module is throwing [a bunch of](https://forum.magicmirror.builders/topic/6851/what-s-casuing-these-ssh-errors-in-the-pm2-logs) SSH related spam into the pm2 error logs.

```
ssh_askpass: exec(/usr/bin/ssh-askpass): No such file or directory
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

I would guess this happens when a modules has a `.git` file, but is not connected to an accessible repo.

fewieden, 2018-04-02:
could also be a private repo you don't have longer access to

MichMich, 2018-04-02:
Yeah, this is caused by a repro to which you don't have access. Wat would be the preferred solution for you? The only thing I can think of is an option to configure which modules should be skipped.

E3V3A, 2018-04-02:
I have no idea what happens, but when I re-instated both `updatenotification` and what I thought to be the offending module (MMM-cryptocurrency) module that I ugly-fixed by renaming its `.git` directory, to `.git_disabled`, the problem is back. YES I can access that repo. 

How can I find what is writing to that log? (at least a PID would have been more helpful.)

MichMich, 2018-04-02:
Could you try removing the module and reinstalling it.

E3V3A, 2018-04-02:
Can you explain in normal `git` language what the `simple-git` npm is doing in that module?

MichMich, 2018-04-02:
It does a git fetch to check what the latest commit is on the remote and compares it to the latest local commit. Using git fetch it can check how many commits the local version is behind the remote version. So for every module that is a git repo, it does a git fetch. If you are no longer allowed to access that repo, you get an error.

It could also be that there is an error with your local git config. By reinstalling the module using a git clone, everything should be reconfigured correctly (or throw an error stating what's wrong).

E3V3A, 2018-04-02:
OMG! This is a fcuking rabbit hole, don't go down it! :scream: 

Ok, this is what I've found so far...

- The error above come from the `openssh-client` binaries, see [1]
- Raspbian OS does **not** come with the `ssh-askpass` apt package, even if it is a recommended add-on to the included `openssh-client`.
- The  `ssh-askpass` apt package is a GUI application to ask for password (and should not be assumed installed)
- Because we're not using a GUI, we need to tell that to ssh. This can be done either by: 
  1. add your `id_rsa.pub` to the server's `~/.ssh/authorized_keys`
  2.  using some kind `--nogui` option to git or ssh
  3. or use some [environment settings in simple-git](https://github.com/steveukx/git-js#environment-variables) 
- The error occurs 2 times at every `updatenotification` trigger (default 10 minutes)
- The error occurs because of the `git fetch` used by the [simple-git](https://github.com/steveukx/git-js) npm package, **but used** in *updatenotification*, in `node_helper.js` here:

https://github.com/MichMich/MagicMirror/blob/develop/modules/default/updatenotification/node_helper.js#L64

So I added a console log: 
```js
    preformFetch() {
        var self = this;
        simpleGits.forEach(function(sg) {
            sg.git.fetch().status(function(err, data) {
                data.module = sg.module;
                if (!err) {
                    console.error("GIT ERROR: git fetch didn't succeed: ", data)
                    sg.git.log({"-1": null}, function(err, data2) {
                        data.hash = data2.latest.hash;
                        self.sendSocketNotification("STATUS", data);
                    });
                }
            });
        });
        this.scheduleNextFetch(this.config.updateInterval);
    },
```

This spits out:

```
GIT ERROR: git fetch didn't succeed:  StatusSummary {
  not_added: [ 'installers/mm2s.sh', 'mm2s.sh' ],
  conflicted: [],
  created: [],
  deleted: [],
  modified:
   [ 'css/main.css',
     'modules/default/alert/alert.js',
     'modules/default/compliments/compliments.js',
     'modules/default/updatenotification/node_helper.js',
     'package-lock.json' ],
  renamed: [],
  files:
   [ { path: 'css/main.css', index: ' ', working_dir: 'M' },
     { path: 'modules/default/alert/alert.js',
       index: ' ',
       working_dir: 'M' },
     { path: 'modules/default/compliments/compliments.js',
       index: ' ',
       working_dir: 'M' },
     { path: 'modules/default/updatenotification/node_helper.js',
       index: ' ',
       working_dir: 'M' },
     { path: 'package-lock.json', index: ' ', working_dir: 'M' },
     { path: 'installers/mm2s.sh', index: '?', working_dir: '?' },
     { path: 'mm2s.sh', index: '?', working_dir: '?' } ],
  ahead: 0,
  behind: 149,
  current: 'master',
  tracking: 'origin/master',
  module: 'default' }

ssh_askpass: exec(/usr/bin/ssh-askpass): No such file or directory
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

```

---

Ref [1]:
```bash
# mygrep "askpass" /usr/bin/ssh*
Binary file /usr/bin/ssh matches
Binary file /usr/bin/ssh-add matches
Binary file /usr/bin/ssh-agent matches
Binary file /usr/bin/ssh-keygen matches

# strings /usr/bin/ssh |grep askpass
/usr/bin/ssh-askpass
ssh_askpass: fflush: %s
internal error: askpass undefined
ssh_askpass: pipe: %s
ssh_askpass: fork: %s
ssh_askpass: dup2: %s
ssh_askpass: exec(%s): %s
```

- The error could also be if when a `.git` doesn't exist, in a `modpath` it will default to the git it finds in a path above it...

This is all I got for this. I've spent way too much time on this tiny issue. 

PS. My *local git config* stuff is working and I don't intend to send or give my password to some modules. (Or even keep it on the RPi for that matter.)

MichMich, 2018-04-02:
Lol. You are really overthinking this. If your OS gives an error you don’t have acces to a file, do you also go look for a stack trace to see which process is responsible for showing user errors? ;)

Anyway: the cause is simple: MagicMirror can’t successfully run the git fetch command since it doesn’t have access to the repository. The repo it is trying to access is probably protected based on your ssh key, which the user on which MagicMirror² runs doesn’t have access to. 

3 possible solutions:
- 1. Allow the mm process to access the repo (read rights). 
- 2. Disable the update notification module. 
- 3. Add a feature to the update notification module which allows you to exclude certain modules.

E3V3A, 2018-04-02:
>  If your OS gives an error you don’t have acces to a file, do you also go look for a stack trace to see which process is responsible for showing user errors?

Ha ha -- eeeh -- Yep, that's what I did! :rofl:

> Allow the mm process to access the repo (read rights).

1. The problem is that I don't know which repo it is! 
2. How would I "give access" to it without giving up ssh credentials? 

TBH, I don't think it is (2) because I can git fetch any repo from command line...

MichMich, 2018-04-02:
The access is probably authorized based on your public ssh key. Not based on credentials. Maybe your MM installation can’t access the public key. 

I suggest you disable all modules and add them one by one to see which one gives the issue. If you found the problematic module(s) try reinstalling them.

E3V3A, 2018-04-02:
~~I have a new idea...~~

:red_circle:    **YES! I got it!**  :1st_place_medal: 

The issue is related to using SSH authentication instead of SSL (HTTPS) for git!
I noticed several repo's (of my own) config files are using the SSH url type:  
`git@github.com:user/repo` instead of the SLL type: `https://github.com/user/repo`.

```bash
# for x in `ls -1d MMM-*`; do echo "check: $x";cat $x/.git/config |grep url; done;
check: MMM-Assistant
        url = git@github.com:eouia/MMM-Assistant
check: MMM-cryptocurrency
        url = https://github.com/matteodanelli/MMM-cryptocurrency.git
```

So you get this spam, any time you have a module with a repo using an SSH url in it's `./.git/config` file. 

I don't know how to check or fix this in the  `updatenotification` module. 
Please make a PR.

MichMich, 2018-04-02:
I use SSH all the time. Works fine as long as your RasPi is configured correctly for the repo’s it tries to access. So no need for a change. 

The logs alerted you of an issue with your config. So they did their job. :)

E3V3A, 2018-04-02:
That is not entirely correct. I'm also using SSH all the time as well, and can do any non-pushing operation without needing credentials. MM should not need any credentials to check for updates. Anything else is a security weakness...even if it doesn't matter that much for us. At the end of the day, I guess one step less for users to have to deal with is better for everyone. (And correct `./.ssh/config` is not that easy for beginners. 

Regardless of what ideology we use, for authentication, the notification module spams the log with useless and cryptic info that nobody can resolve without reading this issue thread.

MichMich, 2018-04-06:
When you use SSH access, MM does not need any credentials. Git does this for you, using your SSH key. 

I agree that the error message is somewhat cryptic, but that's something to solved in the simple-git nom module. Not in MagicMirror. 

The issue is caused by a Raspberry configuration, not by a bug in MagicMirror², therefor I'm closing this issue for now.

closed 2018-04-06 by MichMich