MagicMirrorOrg/MagicMirror #1240: Uncaught TypeError: Cannot read property dismiss of null (alert.js)

E3V3A, 2018-03-31:
I keep getting random events where the error is:  

**`Uncaught TypeError: Cannot read property 'dismiss' of null`**

[image]

The error is coming from the Alert default module on Line 113, in [alert.js](https://github.com/MichMich/MagicMirror/blob/develop/modules/default/alert/alert.js#L113):

```js
	hide_alert: function(sender) {
		//Dismiss alert and remove from this.alerts
		this.alerts[sender.name].dismiss();  // <<========= LINE  113 !
		this.alerts[sender.name] = null;
		//Remove overlay
		var overlay = document.getElementById("overlay");
		overlay.parentNode.removeChild(overlay);
},
```

@MichMich  
Does this have to be `null` or can it also be set to `""` or something else?
Any ideas how to fix or work around this?

I'm surprised nobody else has reported this issue, so I want to eliminate that it is a problem with my module.

E3V3A, 2018-03-31:
I tried to fix with some variations of this:
```js
        //Dismiss alert and remove from this.alerts
//        if (this.alerts[sender.name] != null) {
            this.alerts[sender.name].dismiss();
            this.alerts[sender.name] = "";        // was null
 //       }
```

But it didn't work...and seem to need something else.

MichMich, 2018-04-01:
It's because a alert is dismissed which doesn't exist. Which module is firing the alerts? Might be worth checking that code as well.

E3V3A, 2018-04-01:
It's our module MMM-Assistant, that seem to use Alerts to show pop-up screen results for Snowboy requests... 

But, why didn't my if() statement work then? (Actually it seem to have worked, but then got another error message instead.)

fewieden, 2018-04-02:
`if (this.alerts[sender.name]) {` would cover null, undefined empty string and more

E3V3A, 2018-04-02:
@fewieden  Yeah, but I already tried that before the comments shown above, and it just propagated the error elsewhere... Seem to be some kind of race condition...

fewieden, 2018-04-02:
what is the other error?

I think you need to put 
```
var overlay = document.getElementById("overlay");
overlay.parentNode.removeChild(overlay);
```
if there is no alert to dismiss, there should also be no dom action.
if the overlay is not there, you're trying to access parentNode of null

MichMich, 2018-04-02:
Might be a good idea to check with @paviro, he made the alert module.

paviro, 2018-04-02:
Oh man :D that was ages ago. I will have a look later today :)

MichMich, 2018-04-02:
@paviro It will hunt you for the rest of you life. ;)

paviro, 2018-04-02:
I really hoped someone would replace this 💩  code really fast 😅

MichMich, 2018-04-02:
@E3V3A If that fix seems to work for you, feel free to send a PR. I'll merge it after all test's are finished.

E3V3A, 2018-04-02:
Just tested, @fewieden suggestion fixed it. Now the question is why this happened in the first place. 

Done, while you guys were chatting... ;)
https://github.com/MichMich/MagicMirror/pull/1248

PR #1248 (2018-04-02): null check for notification removal
Make sure there is something to remove, before we attempt to remove the notifications. 
- This fixes #1240
commit 497145b1 (2018-04-02): null check for notification removal

Make sure there is something to remove, before we attempt to remove the notifications. 
- This fixes #1240
closed 2018-04-02 by E3V3A