MagicMirrorOrg/MagicMirror #1243: New update Black Screen

NicoMontanaro, 2018-04-01:
Just updated to the last version of MM, but when i run the script i get just a black blank screen.

kikokazuma, 2018-04-01:
Same problem here after updating this morning, even after a clean install. I was prepping this as a science fair project for my daughter due tomorrow so the timing is absolutely horrible, lol. Is there a way for me to install the previous version?

NPM log shows the following:

Starting MagicMirror: v2.3.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
All module helpers loaded.
Starting server on port 8080 ...
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Whoops! There was an uncaught exception...
{ Error: listen EADDRINUSE 127.0.0.1:8080
    at Object.exports._errnoException (util.js:1050:11)
    at exports._exceptionWithHostPort (util.js:1073:20)
    at Server.setupListenHandle [as _listen2] (net.js:1263:14)
    at listenInCluster (net.js:1304:12)
    at doListen (net.js:1428:7)
    at GetAddrInfoReqWrap.asyncCallback [as callback] (dns.js:62:16)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:78:10)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '127.0.0.1',
  port: 8080 }
MagicMirror will not quit, but it might be a good idea to check why this happene                                                                                                                                                             d. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://gi                                                                                                                                                             thub.com/MichMich/MagicMirror/issues
Launching application.

kikokazuma, 2018-04-01:
Dear 8 pound, 6 oz baby Jesus, save me.

MichMich, 2018-04-01:
It seems the port is in use: `EADDRINUSE 127.0.0.1:8080` Did you try rebooting you Pi?

kikokazuma, 2018-04-01:
Yes, I've rebooted several times and even tried a fresh install of Raspbian + MagicMirror.

kikokazuma, 2018-04-01:
Was working fine last night and broke after this morning's update, much like the OP.

NicoMontanaro, 2018-04-01:
I did, but still no answer. Thanks!

MichMich, 2018-04-01:
What commands did you guys use to update?

hermannw, 2018-04-01:
Same problem here.
mm-error-0.log contains these lines several hundred times:

TypeError: Cannot read property 'hash' of null
    at /home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:68:31
    at Git.<anonymous> (/home/pi/MagicMirror/node_modules/simple-git/src/git.js:1482:10)
    at done (/home/pi/MagicMirror/node_modules/simple-git/src/git.js:1354:21)
    at ChildProcess.<anonymous> (/home/pi/MagicMirror/node_modules/simple-git/src/git.js:1379:16)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

and then at the end:

npm ERR! Linux 4.4.43-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v6.12.3
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.3.0 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.3.0 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.

mm-out-0.log on the other hand says everything is running ok...

/home/pi/.pm2/logs/mm-out-0.log                                                                                                               166857/163K              100%
> sh run-start.sh

Starting MagicMirror: v2.3.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
Initializing new module helper ...
Module helper loaded: newsfeed
No helper found for module: MM-Remote-Control.
Initializing new module helper ...
Module helper loaded: mmm-weatherchart
Initializing new module helper ...
Module helper loaded: MMM-Tools
No helper found for module: worldclock.
No helper found for module: MMM-CurrencyExchange.
Initializing new module helper ...
Module helper loaded: MMM-NOAA
Initializing new module helper ...
Module helper loaded: MMM-ImagesPhotos
All module helpers loaded.
Starting server on port 8080 ...
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Connecting socket for: mmm-weatherchart
Starting node helper: mmm-weatherchart
Connecting socket for: MMM-Tools
Connecting socket for: MMM-NOAA
Starting module: MMM-NOAA
Connecting socket for: MMM-ImagesPhotos
Starting node helper for: MMM-ImagesPhotos
Sockets connected & modules started ...
Launching application.

Any help would be appreciated.

kikokazuma, 2018-04-01:
As per instructions on https://github.com/MichMich/MagicMirror#updating

I ran this from the MagicMirror directory:

git pull && npm install

hermannw, 2018-04-01:
Michael,

I used the recommended git pull && npm install.

NicoMontanaro, 2018-04-01:
git pull && npm install

MichMich, 2018-04-01:
Try disabling all non-default modules.

kikokazuma, 2018-04-01:
Hi MichMich

Given I am running a clean install, would there by any non-default modules?

MichMich, 2018-04-01:
@kikokazuma Not by default. - Any messages in your logs?

kikokazuma, 2018-04-01:
@MichMich pardon my noobishness, is there a log I can provide other than the NPM log I posted above? If so, what command can I run to help?

Thanks in advance!

hermannw, 2018-04-01:
Nope. Disabling modules did not help.

MichMich, 2018-04-01:
AH, sorry, no that was what I was looking for (only noticed @hermannw 's log). 
@kikokazuma You log states you are already running something on port 8080. Are you sure you're not running two instances at the same time?

MichMich, 2018-04-01:
@hermannw Try disabling the updatenotification module.

hermannw, 2018-04-01:
Nope. No change. I disabled everything except the clock.

kikokazuma, 2018-04-01:
@MichMich Previously, if I ran sudo netstat -lnptu it would tell me electron was already running on that port.

Strangely, just one minute ago, I rebooted and it seems to be working again.
Others reported similar issues: https://forum.magicmirror.builders/topic/6874/so-to-update-magicmirror-i-would

Something is/was going on.

MichMich, 2018-04-01:
So, just to be clear @kikokazuma, it is working for you now? If so, what version of Node are you running?

Hurg84, 2018-04-01:
Hi MichMich, hi other Magicmirror Users,

I can confirm that after the upgrade the physical display stays black,
Reinstalling eletron doesnt fix the problem.

On the other hand ip:8080 in a Browser works, so it's just the mirror's browser.

Regards

Hurg

desq42, 2018-04-01:
Hi,

same symptoms here as Hurg84. Applications works remotely, but die Screen stays black.
Node -v -> v6.14.1

Regards

desq

kikokazuma, 2018-04-01:
@MichMich correct. I rebooted again and it is still working for me.

If I run node -v, I get:
v9.10.1

Hurg84, 2018-04-01:
I got node v6.14.1 like desq42

MichMich, 2018-04-01:
Could someone try the following in the MagicMirror folder:
```
npm install electron@1.4.15
```

If this works, please state your Node version as well. (`node -v`)

Hurg84, 2018-04-01:
@MichMich It works with node v6.14.1

MichMich, 2018-04-01:
So an electron downgrade solved it for you?

Hurg84, 2018-04-01:
Yes, I can confirm that it works fine now.

MichMich, 2018-04-01:
Ok, I'll release a new version with a downgraded electron.

NicoMontanaro, 2018-04-01:
Worked for me! node version 6.14.15

MichMich, 2018-04-01:
https://github.com/MichMich/MagicMirror/releases/tag/v2.3.1

E3V3A, 2018-04-01:
Hi Guys! 

@Hurg84 
> It works with node v6.14.1
@MichMich 
> Ok, I'll release a new version with a downgraded electron.

IMHO, this doesn't make sense at all. You're updating everything except the most relevant thing, which is node itself. Can you please explain the reasoning behind this?

MichMich, 2018-04-01:
You mean electron? Node isn't upgraded.

Hurg84, 2018-04-01:
Guess he's refering to solving the problem by upgrading node into a version that works with the new electron version instead ob downgrading electron to a node compatible version.

MichMich, 2018-04-01:
I upgraded node, but that didn’t seem to solve the issue. Downgrading electron is an intermediate solution until we find a way to get the new electron version to work.

yatmanyatan, 2018-04-01:
@MichMich  v9.10.1 electron-downgrade is solved my problem too thx

E3V3A, 2018-04-01:
> I upgraded node, but that didn’t seem to solve the issue. Downgrading electron

Yeah, but you never actually upgraded electron fully, only to `1.7.13`, when the [latest](https://github.com/electron/electron/releases) is `1.8.4`.  Then again I have no idea how those guys do their versioning there. They have a 1.8.3 released before the 1.7.13... However, there were some bug fix mentions of windows not showing in the 1.8.4.

MichMich, 2018-04-01:
Nope. Tried 1.8.4 as well (using Node 9).

ezar, 2018-04-01:
I try electron 1.8.4 using node 9, fails 👎

justjim1220, 2018-04-01:
The developer version 2.4.0 did not have these issues...

justjim1220, 2018-04-01:
Electron 1.8.4 has been giving a lot of developers problems. Bugs need worked out. Thus the reason 1.7.3 is still available and widely used. 

Funny thing, you say it's an issue with Windows, yet I am running Windows 10, node 9.9.0, & electron 1.8.4 as well as the developer MM 2.4.0 and it fan perfectly the first time I tried. 

Messed up a little now though, trying to put same tweaks I was running with last version and have a syntax error somewhere... SMH...

E3V3A, 2018-04-02:
@desq42 
> same symptoms here as Hurg84. Applications works remotely, but die Screen stays black.
> I can confirm that after the upgrade the physical display stays black, 
> On the other hand ip:8080 in a Browser works, so it's just the mirror's browser.

This is interesting. Can you guys please re-post the log messages? Just the error plus a few lines before & after the error should suffice. And disable all mods except clock.

Hurg84, 2018-04-02:
Hi EVA,

I'm not yet that familiar with the Mirrorsoftware yet.
Where are the logs located?

ezar, 2018-04-02:
pm2 logs

MichMich, 2018-04-02:
@E3V3A It's obviously an issue with electron. The server is just a simple node (express) server which doesn't need electron to start. So if the server responds, and electron doesn't work, the issue is caused by electron. This is why I tried downgrading electron, which 'solved' the issue (for the time being). 

Apparently, Electron is having a hard time running on the Pi platform. We will need to investigate a way to get a more recent version of Electron up and running. A pull request for this would be very welcome. It might be a good idea to start a new branch for this.

I'll reopen this issue so we can investigate this further.

VH-Tech, 2018-04-02:
Hi, downgrading the electron solved the problem for me as well!!

E3V3A, 2018-04-02:
@MichMich  Unless people actually **start posting the errors they get, and how they got them in the first place**, there is not much we can do. Just hearing that they *got a black screen* or it *doesn't work*. just isn't helping anyone. 

Like @Kiina mentioned in another issue, apparently there was no errors when people (who?) were running the dev editions... Or maybe nobody ever bothered to test the dev?? 

Either way, when telling people to update with "git pull" etc, I guess we also need to tell them to also update all their OS, apt packages and what not, causing a bunch of irrelevant error reports.

MichMich, 2018-04-02:
My guess is no one is running the dev branch. 

Or -just like me- forget to run npm install after the git pull. (This is the reason why I didn’t notice the issue with Electron after we bumped the version. ☺️)

E3V3A, 2018-04-02:
@VH-Tech 
> Hi, downgrading the electron solved the problem for me as well!!

Hey! Downgrading is cheating! Don't downgrade and help us out debugging. :D :rofl:

Kiina, 2018-04-02:
@E3V3A well not everyone wants to try debugging and solving the issue, esp if they just want to run the mirror. But maybe the OS would be interesting. Is it really a issue with the raspberry or maybe Raspbian? Anyone tried a different OS?

MichMich, 2018-04-02:
Nope. It’s an issue with electron.

Kiina, 2018-04-02:
Well the thing it it works fine on Windows and Linux on non rpi devices. And I don't see anything general about electron being broken on rpi since other projects seem to be working just fine. So there has to be an issue with some implementations or weird combinations. Also I'm kinda sure it was working one point in time but I will look into it when I got a few more spare rpi and sd cards

Hurg84, 2018-04-03:
Sorry, I'm late.

---mm-error-0.log after upgrade to 2.3.0---

npm ERR! Linux 4.14.31-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v6.14.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.3.0 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.3.0 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
(node:991) DeprecationWarning: sys is deprecated. Use util instead.
(node:1112) DeprecationWarning: sys is deprecated. Use util instead.

--END--

---mm-out-0.log after upgrade to 2.3.0---

> magicmirror@2.3.0 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.3.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
Initializing new module helper ...
Module helper loaded: MMM-dht22
All module helpers loaded.
Starting server on port 8080 ... 
You're using a full whitelist configuration to allow for all IPs
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Connecting socket for: MMM-dht22
Starting node helper: MMM-dht22
Sockets connected & modules started ...
Launching application.
Create new calendar fetcher for url: http://p04-calendarws.icloud.com/ca/subscribe/1/MvNZsi9P7yIjDuxhTql8JuqzA-OnNL6inOjcNv-7-qE7DC_Xs-fglQvRqSwngzdA - Interval: 300000
Create new calendar fetcher for url: http://ifeiertage.de/nw-s.ics - Interval: 300000
Create new news fetcher for url: https://www.n-tv.de/rss - Interval: 300000

---END---

Hope this helps in any way.

winstonma, 2018-04-04:
Hi @MichMich I just updated MagicMirror to v2.3.1 on my RPi2.

I used the default config.js (without any custom addon) and it still blank screen.

justjim1220, 2018-04-04:
I copy paste to replace the new config.hs with my old one. Did same with
package.json and package-lock Jon.

Then did npm rebuild.

Now it works!

justjim1220, 2018-04-04:
With any of the default modules… If you made any changes to the .js files or the README.md files, AND, you would like to keep those changes… TRY THIS:

cut and paste any of the default modules from the MagicMirror directory to any other directory.
You also need to do the same for the package.json, package-lock.json, and your config.js file if there are any additional modules that have been added.

then do the ‘git pull $$ npm install’

when it is done, replace those files in the new MagicMirror that just got installed with the ones you saved from your old MagicMirror.

then do ‘npm rebuild’

It took a couple of times starting the MagicMirror to get all the changes to take…

this also fixes the currentweather module problem…

and the blank screen problem…

Hurg84, 2018-04-05:
@MichMich I've managed to update/upgrade anything but electron.

I keep getting the following error:

---Start---
/home/pi/MagicMirror/node_modules/electron/install.js:47
  throw err
  ^

Error: Could not parse checksum file at line 1: 
    at new ChecksumParseError (/home/pi/MagicMirror/node_modules/sumchecker/build.js:71:127)
    at /home/pi/MagicMirror/node_modules/sumchecker/build.js:130:20
    at Array.forEach (<anonymous>)
    at /home/pi/MagicMirror/node_modules/sumchecker/build.js:125:38
    at new Promise (<anonymous>)
    at ChecksumValidator.parseChecksumFile (/home/pi/MagicMirror/node_modules/sumchecker/build.js:121:14)
    at <anonymous>
---End---

I tried so previous fixes from older electron version and Parameters like "--unsafe-perm=true --allow-root" without any success.

After this error either Electron immediately crashes or remains a black screen and the webserver doesnt respond to external Brwosers either.

Good thing is in this state running "git pull && npm install" in the Magicmirror directory followed by a reboot brings to Mirror back to life.

I'll try upgradeing electron the following days again.
I'd aprechiate helpful replies :)

Regards

Hurg

Hurg84, 2018-04-05:
hi *, me again :)

If anyone wants to try electron too, here’s what i’ve done so far and my Mirror is still working:

Updating Note:

sudo curl -sL https://deb.nodesource.com/setup_9.x | sudo bash -
sudo apt-get install -y nodejs

Updating npm:

sudo npm install npm@latest -g

Installing Prerequisites for Electron beta:

sudo npm install grunt@latest -g
sudo npm install grunt-eslint@latest -g
sudo npm install grunt-yamllint@latest -g

If your Mirror crashes immediately after trying to install electron@beta
or remains a black screen without any sign of life even with a remote browser run:

git pull && npm install

in the Magicmirror directory followed by a reboot.

winstonma, 2018-04-06:
@justjim1220 I tried reverting `package.json` and `package.json.lock`, remove the whole `node_modules `folder and run `npm install && npm start`. But still no luck.

@MichMich Thanks for the info. I updated the electron version to the latest beta and now it works.

[52 more replies]

closed 2018-06-27 by MichMich