MagicMirrorOrg/MagicMirror #1247: currentweather.js fails to display Fahrenheit

pdanielsen, 2018-04-01:
**Platform**: Raspberry Pi Zero W with raspbian-stretch 9.1, chromium-browser 60.0.3112.89

**Node Version**: v6.14.1

**MagicMirror Version**: v2.3.1

**Description**: When config/config.js has units = "imperial", the currentweather module doesn't appear at all.  This is due to a typo in currentweather.js.  The variable 'windChillInF' is referred to at line 417, but doesn't exist.  It was declared and calculated at line 410 as 'windchillinF'. 

Change the declaration and calculation to 'windChillInF' to follow the convention for the 'windChillInC' variable used for Celsius.

**Steps to Reproduce**: 
In config/config.js
  Set units = "imperial"
  Include the currentweather module in the "modules" list
Start the mirror software
Visit the mirror server in a browser

**Expected Results**: The currentweather module should appear in the browser.

**Actual Results**: The currentweather module did not appear in the browser.

**Configuration**: 
...
var config = {
  ...
  units: "imperial",
  ...
  modules: [
    {
    module: "currentweather",
    position: "top_right",
    config: {
      location: "New York",
      locationID: "5128581",  //ID from http://www.openweathermap.org/help/city_list.txt
      appid: "YOUR_OPENWEATHER_API_KEY"
    },
    ...
  ]
  ...
}

MichMich, 2018-04-02:
I made a change which should solve this: https://github.com/MichMich/MagicMirror/commit/10eb41d31943a4495bf4c4674d52d0b9631a7446

Could you give the develop branch a try and let me know if this fixed your issue?

CraigDamlo, 2018-04-02:
I haven't tried the developer branch, but can confirm on the main branch. The forecast works fine, just not the current.

pdanielsen, 2018-04-02:
> I made a change which should solve this: 10eb41d <https://github.com/MichMich/MagicMirror/commit/10eb41d31943a4495bf4c4674d52d0b9631a7446>

Yes. That fixed it.

Thanks for the quick turn-around!

closed 2018-04-02 by MichMich